tinygo-org/tinygo #1323: exec: "avr-gcc": executable file not found in %PATH%

RensTillmann, 2020-08-27:
Trying to flash Arduino Uno with tinygo. I followed this guide: [https://tinygo.org/getting-started/windows/](https://tinygo.org/getting-started/windows/)

I have added C:\tinygo\bin to the %PATH%
But when I run the following command I get this error:
```
$ pwd
~/go/src/github.com/RensTillmann/6dof
$ tinygo flash -target arduino main.go
error: failed to build src/runtime/scheduler_avr.S: exec: "avr-gcc": executable file not found in %PATH%
```

RensTillmann, 2020-08-27:
Additional info:
When I run the `tinygo env` the output is as follows:

```
tinygo env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:\\Go"
GOPATH="C:\\Users\\Rens\\go"
GOCACHE="C:\\Users\\Rens\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\tinygo"
```

RensTillmann, 2020-08-27:
Versions:

```
go version go1.15 windows/amd64
tinygo version 0.14.1 windows/amd64 (using go version go1.15 and LLVM version 10.0.1)
```

aykevl, 2020-08-27:
This means you need to install avr-gcc.

I do not have experience with this myself, perhaps someone else has?

sago35, 2020-08-27:
I don't have an AVR, so I've never tried it.
But I was able to build it by following these steps

1. install AVR 8-bit Toolchain v*.** – Windows
  * https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers
2. Unzip the file and pass the path into the bin folder.

```
C:\tinygo>set PATH=C:\tinygo\avr8-gnu-toolchain-3.6.2.1778-win32.any.x86\avr8-gnu-toolchain-win32_x86\bin;%PATH%

C:\tinygo>where avr-gcc
C:\tinygo\avr8-gnu-toolchain-3.6.2.1778-win32.any.x86\avr8-gnu-toolchain-win32_x86\bin\avr-gcc.exe

C:\tinygo>tinygo version
tinygo version 0.14.0 windows/amd64 (using go version go1.14.3 and LLVM version 10.0.1)

C:\tinygo>tinygo build -size short -o test.hex -target=arduino examples/serial
   code    data     bss |   flash     ram
    670      82     658 |     752     740
```

sago35, 2020-08-27:
You probably need avrdude as well.
I haven't tried it, though it seems to be possible to install it from the following sources

http://download.savannah.gnu.org/releases/avrdude/
* avrdude-6.3-mingw32.zip

alankrantas, 2020-08-28:
I use this on Windows 10:

[AVR-GCC for Windows 32 and 64 bit] (https://blog.zakkemble.net/avr-gcc-builds/)

Unzip it to C:\ and add C:\avr-gcc-x.x.x-x64-mingw\bin to PATH.

thushan, 2020-09-16:
I've added support for [avr-gcc on scoop](https://github.com/ScoopInstaller/Main/pull/1400#pullrequestreview-486490063) should be merged to master in a couple of days.

> scoop install avr-gcc

Hope that helps folks - additionally ]TinyGo is already there](https://github.com/ScoopInstaller/Main/pull/1390)!

sago35, 2021-02-28:
I don't have an AVR, so I haven't tried it, but I think the following is the easiest.
I think it needs to be added to tinygo-site.

```
$ scoop install avr-gcc
$ scoop install avrdude
```

thushan, 2021-02-28:
Agreed @sago35, will update doc site to reflect that.

Gustavomurta, 2021-05-29:
I have problem when installing avr-gcc. 

scoop install avr-gcc
Installing 'avr-gcc' (10.2.0) [64bit]
The URL is not valid : 
URL https://dl.bintray.com/lumito/public/projects/avr-gcc/10.2.0/avr-gcc-10.2.0-x64-windows.7z is not valid

I found it at this site:
https://avr.lumito.net/downloads/
https://avr.lumito.net/download/4944/

Please,update this procedure. 
Thanks.

thushan, 2021-05-29:
Thanks for the heads up @Gustavomurta, will look into it and get back to you soon. Looks like the URL patterns have changed by that URL, but will confirm properly.

thushan, 2021-05-30:
> I have problem when installing avr-gcc.
> 
> scoop install avr-gcc
> Installing 'avr-gcc' (10.2.0) [64bit]
> The URL is not valid :
> URL https://dl.bintray.com/lumito/public/projects/avr-gcc/10.2.0/avr-gcc-10.2.0-x64-windows.7z is not valid
> 
> I found it at this site: (avr-gcc-11-20210501-x64-windows.7z)
> https://avr.lumito.net/downloads/
> https://avr.lumito.net/download/4944/
> 
> Please,update this procedure.
> Thanks.
@Gustavomurta 
Can you open up an [issue on scoop](https://github.com/lukesampson/scoop/issues) for me to address the package issue... if there's an error message that appears that would be helpful too. I don't have a Windows machine to check it right at this moment but will do so tomorrow.

thushan, 2021-06-01:
Reverted to the official builds by Zak, tested and opened a PR. Should go through in a few days @Gustavomurta.

thushan, 2021-09-30:
https://github.com/ScoopInstaller/Main/pull/2246 was merged to master, so you should be able to run avr-gcc properly now :)

aykevl, 2022-11-18:
The dev branch doesn't need avr-gcc anymore, so this has been fixed.

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing. Thank you!

closed 2023-02-12 by deadprogram