tinygo-org/tinygo #1327: [runtime error: out of memory] due to repeated call to demo splitter method

Kusers, 2020-08-29:
Error screenshot
https://github.com/Kusers/tinygoDemo/blob/master/runerr.png
javascript code
https://github.com/Kusers/tinygoDemo/blob/master/run_err_js.txt
How can we solve this problem? I look forward to your reply. Thank you very much

aykevl, 2020-08-29:
Passing data between JavaScript and WebAssembly (Go) currently leaks memory. You can slow it down a bit with the `-heap-size` parameter.

Kusers, 2020-08-31:
If I use C language, will it still happen?

deadprogram, 2020-09-12:
@Kusers this is a really a general problem with memory management and any WASM code that wants to pass data to/from JavaScript, regardless of what was used to compile it.

FrankReh, 2020-10-14:
The memory leak caused by having no finalizers called? I've run into that today.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram