tinygo-org/tinygo #1330: make make-smoketest faster

sago35, 2020-08-30:
I've been using make smoketest a lot lately, but it's very slow and it's a pain to check.
I would like to speed up the process by doing one of the following.
Or do you have any other ideas?

1. Separate targets for speed with make -j
    * https://github.com/tinygo-org/tinygo/commit/3bd1878832e376f47fc65d26e42bf5f04fd58bd5
    * 48 sec (make smoketest) -> 19 sec (make smoketest -j --output-sync)
2. Use another build tool, such as ninja.
3. Use go script to perform a fast build check.

note: I would like to speed up CI as well. But the speed of CI depends on the number of CPUs.

sago35, 2020-09-05:
| os | before | make-j | go sctipt |
|-----|------------|----------|-----------|
| linux | 106 s | 41 s | 29 s |
| macos | 172 s | 59 s | 49 s |
| windows | 389 s | 102 s | |

deadprogram, 2024-12-07:
@sago35 can we close this specific issue now?

sago35, 2024-12-07:
I will investigate this issue again.

Adding something like the following, as done in `drivers`, might help reduce CI times. 

https://github.com/tinygo-org/drivers/blob/v0.29.0/smoketest.go

davecheney, 2026-10-26:
Thanks for raising this. The smoke tests are now split into independent groups that can run in parallel with `make -j smoketest`.

On my macOS arm64 machine, using current `dev` (`bcbb86e5`) with pinned LLVM 22.1.4, both runs passed:

| Command | Elapsed time |
|---|---:|
| `make smoketest` | 235.56 seconds |
| `make -j smoketest` | 109.63 seconds |

That is about 2.15× faster. Both runs used the same compiler and retained build cache; LLVM and TinyGo build time was excluded.