tinygo-org/tinygo #1337: docker: install LLVM version that allows for our Xtensa support for tinygo-dev

deadprogram, 2020-08-31:
We will need to install a version of LLVM in order to build the tinygo-dev Docker container in a way that includes this support:
https://github.com/tinygo-org/tinygo/blob/dev/Dockerfile#L4-L7

aykevl, 2020-08-31:
I do not think Espressif provides binaries yet, see https://github.com/espressif/llvm-project/issues/36 for example.
I'm not sure whether compiling it from source for Docker is feasible, to be honest.

deadprogram, 2020-09-09:
Valid point. If the Espressif people setup a build server with binaries, that would be ideal.

If not, perhaps we should setup our own build server perhaps as a separate docker container/stage so build times could improve.

deadprogram, 2021-04-22:
Closing since this is not really an issue so much with how we are doing it.

closed 2021-04-22 by deadprogram