tinygo-org/tinygo #1339: Blue Pill Interrupt on GPIO input state change

olablt, 2020-08-31:
I have Blue Pill STM32f103C8T6

How to create Interrupts on 3 GPIO inputs, to fire after each state change?
Is it possible?

Here is an example Arduino code using interrupts. I am not sure how to do this with TinyGo
https://github.com/chrisalbertson/quadratureBluePill/blob/master/quadratureBluePill.h

ofauchon, 2021-01-10:
Hi. 

I can't find implementation of (Pin)SetInterrupt for STM32 in the Tinygo sources. 

So I guess GPIO interrupts are not yet available for STM32. 

However, you can find some example of low-level code to configure GPIO interrupts  : 

- on STM32L0: https://github.com/ofauchon/tinygo/blob/lgt92_examples/src/examples/lgt92/01_button_interrupt.go
- on STM32F103: https://github.com/ofauchon/experiments/blob/master/tinygo/bluepill-misc/test_gpio_int.go

Hope that helps

Olivier

kenbell, 2021-07-09:
Closing since GPIO interrupt support is merged

closed 2021-07-09 by kenbell