tinygo-org/tinygo #1340: Error : flash on Arduino Mega 2560

NLGRF, 2020-09-01:
## Code :
```go
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 1000)

		led.High()
		time.Sleep(time.Millisecond * 1000)
	}
}
```
## Using :
```cmd
tinygo flash -target arduino-mega2560 -port COM4 .
```

## I was getting:
```
avrdude: invalid file format '\Users\non19\AppData\Local\Temp\tinygo411124079\main.hex' in update specifier
avrdude: error parsing update operation 'flash:w:C:\Users\non19\AppData\Local\Temp\tinygo411124079\main.hex'
error: failed to flash C:\Users\non19\AppData\Local\Temp\tinygo411124079\main.hex: exit status 1
```

🔢 I try to flash on the Arduino UNO, Arduino Nano by arduino, arduino-nano. => `Success` 👍

aykevl, 2020-09-02:
I think I see the problem.

Can you try editing the file `targets/arduino-mega2560.json`?

```diff
-    "flash-command":"avrdude -c wiring -b 115200 -p atmega2560 -P {port} -U flash:w:{hex} -v -D"
+    "flash-command":"avrdude -c wiring -b 115200 -p atmega2560 -P {port} -U flash:w:{hex}:i -v -D"
```

The difference is the extra `:i` at the end.

aykevl, 2020-09-02:
See: https://github.com/tinygo-org/tinygo/pull/1341

NLGRF, 2020-09-02:
> See: #1341

> I think I see the problem.
>
> Can you try editing the file `targets/arduino-mega2560.json`?
>
> ```diff
> -    "flash-command":"avrdude -c wiring -b 115200 -p atmega2560 -P {port} -U flash:w:{hex} -v -D" [...]

@aykevl Thanks

NLGRF, 2020-09-02:
> I think I see the problem.
>
> Can you try editing the file `targets/arduino-mega2560.json`?
>
> ```diff
> -    "flash-command":"avrdude -c wiring -b 115200 -p atmega2560 -P {port} -U flash:w:{hex} -v -D" [...]

what is `:i` , stands for?

deadprogram, 2020-09-18:
This issue was corrected in v0.15.0 which was just released. Now closing.

closed 2020-09-18 by deadprogram