tinygo-org/tinygo #1352: compiler: Support Go 1.16 file embedding

bgould, 2020-09-04:
Assuming that https://github.com/golang/go/issues/41191 is adopted, presumably support will need to be added to the compiler in order to support this feature.  Among other things, it seems like this could be a reliable means of defining byte arrays stored in flash on microcontrollers (with the current mechanism, it can be unpredictable at times, and difficult to determine why some variables are determined to be constants and some are not).

aykevl, 2020-09-04:
Yes, I really want this. However I'm waiting at least until the proposal is accepted before doing anything.

> eventually once a beta implementation in Go 1.16 is available we can evaluate if any special work is necessary to support the new functionality.

This will certainly require some custom work in TinyGo. Ideally it should work together with dead code elimination so that only files that are absolutely needed are included.

ncw, 2021-03-08:
go1.16 is out now with fully baked [embed](https://golang.org/pkg/embed/) support.

Now that tinygo 0.17.0 is released with go1.16 support I just noticed that this doesn't work when trying to compile a wasm binary.

It would be a great feature :-)

ii64, 2022-06-01:
Hi, is this resolved by https://github.com/tinygo-org/tinygo/pull/2873 ?

ncw, 2022-06-08:
@ii64 yes it looks like that does resolve it :-)

It isn't released yet but assuming it makes it into the master branch it should be in 0.24.0 :crossed_fingers:

aykevl, 2022-06-08:
Yes, unless there is a big showstopper (that I don't expect) this will definitely be part of the next release :)

> This will certainly require some custom work in TinyGo. Ideally it should work together with dead code elimination so that only files that are absolutely needed are included.

This works in #2873. The linker can even merge files that happen to have the same contents, and `//go:embed` for byte slices that don't have their contents modified. And `//go:embed` globals that aren't used won't be part of the resulting binary.

closed 2022-07-01 by deadprogram