tinygo-org/tinygo #1358: Constants, Functions etc. from "machine/board_*" @IDE (Goland)

hbroer, 2020-09-05:
Hi,

I checked out TinyGo yesterday and noticed a problem that all constants from the `machine/board*` files are not available. I am using the Feather M0 so it's a samd21 chip. How can I enable that the Pin constants for example are available in my IDE?

This is an error in Goland, but it works when building it:
```
machine.Pin(machine.PA16)
```

This works with Goland and the build:
```
machine.Pin(16)
```

aykevl, 2020-09-05:
Hi @hbroer, this is not yet fully supported but some people have had success with this setup: https://tinygo.org/ide-integration/

hbroer, 2020-09-05:
Hi,

yea I found that. It only works for the generic functions, constants etc. in machine like `machine.Pin`, but not for the device specific objects like `machine.PA16`.

I am not sure if this even can be fixed by TiniGo because of its nature, or if that has to be added as a feature by Jetbrains to their IDE. If so I would open a Issue there but first I wanted to ask here. ;)

aykevl, 2020-09-05:
What the documentation doesn't cover is how you can set the right build tags. TinyGo uses a lot of build tags to select the right board/chip, which you can find with `tinygo info <target>` such as `tinygo info arduino-nano`.

hbroer, 2020-09-05:
I have no problem to build it, but my IDE complains about that for example `machine.PA16` does not exist. And I have no intellisense for the device/board specific objects. I found the `PA16` only by looking into the sources and was surprised that `machine.PA16` did build. The IDE only finds `machine_generic.go` and `machine.go`. The IDE is not aware of `machine_atsamd21.go`.

Like I said I am not sure if this can even fixed by TinyGo, or if I have to open a issue @ Jetbrains to ask to support TinyGo.

Btw great work dude, this is an awesome project with great potential to take over the world ;)

sago35, 2020-09-05:
The following should help you

* https://github.com/tinygo-org/tinygo-site/pull/107
    * https://deploy-preview-107--tinygo.netlify.app/ide-integration/

hbroer, 2020-09-05:
@sago35 oh thank you so much. That's new, I have the old version still open in another browser tab :D I will test it tomorrow and give feedback. 👍

hbroer, 2020-09-06:
I got it to work with the help of the updated integration page mentioned by @sago35 and added how I did on windows 10 with Jetbrains Goland here: https://github.com/tinygo-org/tinygo-site/pull/107#issuecomment-687781692

closed 2020-09-06 by hbroer