tinygo-org/tinygo #1359: How do Unit Testing work with TinyGo

hbroer, 2020-09-05:
Hi,

I did not found information on how unit tests work with TinyGo. I tried it with `tinygo test pathto_test.go` but it fails. Where can I find the information on how it works?

```
tinygo test fan/fan_test.go
# runtime
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:29:21: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:80:42: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\gc_extalloc.go:412:4: undeclared name: extfree
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\gc_extalloc.go:577:10: undeclared name: extalloc
C:\Go\src\runtime\panic.go:13:2: undeclared name: abort
C:\Go\src\runtime\panic.go:20:2: undeclared name: abort
C:\Go\src\runtime\print.go:14:3: undeclared name: putchar
C:\Go\src\runtime\print.go:26:3: undeclared name: putchar
C:\Go\src\runtime\print.go:35:4: undeclared name: putchar
C:\Go\src\runtime\print.go:58:3: undeclared name: putchar
C:\Go\src\runtime\print.go:68:3: undeclared name: putchar
C:\Go\src\runtime\print.go:89:3: undeclared name: putchar
C:\Go\src\runtime\print.go:95:3: undeclared name: putchar
C:\Go\src\runtime\print.go:183:3: undeclared name: putchar
C:\Go\src\runtime\print.go:188:2: undeclared name: putchar
C:\Go\src\runtime\print.go:195:2: undeclared name: putchar
C:\Go\src\runtime\print.go:202:2: undeclared name: putchar
C:\Go\src\runtime\print.go:206:2: undeclared name: putchar
C:\Go\src\runtime\print.go:207:2: undeclared name: putchar
C:\Go\src\runtime\print.go:253:3: undeclared name: putchar
C:\Go\src\runtime\print.go:262:3: undeclared name: putchar
C:\Go\src\runtime\print.go:264:3: undeclared name: putchar
C:\Go\src\runtime\print.go:275:2: undeclared name: putchar
C:\Go\src\runtime\print.go:283:2: undeclared name: putchar
C:\Go\src\runtime\print.go:284:2: undeclared name: putchar
C:\Go\src\runtime\print.go:288:4: undeclared name: putchar
C:\Go\src\runtime\print.go:290:4: undeclared name: putchar
C:\Go\src\runtime\runtime.go:64:9: undeclared name: ticksToNanoseconds
C:\Go\src\runtime\runtime.go:64:28: undeclared name: ticks
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:88:9: undeclared name: ticks
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:118:10: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:123:10: undeclared name: ticks
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:128:53: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:131:26: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:140:8: undeclared name: asyncScheduler
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:146:16: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:150:39: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:153:4: undeclared name: sleepTicks
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:154:7: undeclared name: asyncScheduler
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler_any.go:10:31: undeclared name: nanosecondsToTicks
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler_any.go:20:3: undeclared name: postinit
```

--help does not help ^^

```
tinygo test --help
Usage of tinygo:
  -cflags string
        additional cflags for compiler
  -dumpssa
        dump internal Go SSA
  -gc string
        garbage collector to use (none, leaking, extalloc, conservative)
  -heap-size string
        default heap size in bytes (only supported by WebAssembly) (default "1M")
  -ldflags string
        additional ldflags for linker
  -no-debug
        disable DWARF debug symbol generation
  -o string
        output filename
  -ocd-output
        print OCD daemon output during debug
  -opt string
        optimization level: 0, 1, 2, s, z (default "z")
  -panic string
        panic strategy (print, trap) (default "print")
  -port string
        flash port
  -print-stacks
        print stack sizes of goroutines
  -printir
        print LLVM IR
  -programmer string
        which hardware programmer to use
  -scheduler string
        which scheduler to use (none, coroutines, tasks)
  -size string
        print sizes (none, short, full)
  -tags string
        a space-separated list of extra build tags
  -target string
        LLVM target | .json file with TargetSpec
  -verifyir
        run extra verification steps on LLVM IR
  -wasm-abi string
        WebAssembly ABI conventions: js (no i64 params) or generic (default "js")
```

aykevl, 2020-09-05:
`tinygo test` is not fully supported yet. I've been working on it but it doesn't yet work as well as `go test`.

Depending on your package, you could simply use `go test` instead.

```
tinygo test fan/fan_test.go
# runtime
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:29:21: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\scheduler.go:80:42: undeclared name: timeUnit
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\gc_extalloc.go:412:4: undeclared name: extfree
..\..\..\..\AppData\Local\tinygo\goroot-go1.14.6-49110b6531fbc51373ef78b6542b7c9d9fe63d55ee73fcf9fbaefedf81e90290\src\runtime\gc_extalloc.go:577:10: undeclared name: extalloc
C:\Go\src\runtime\panic.go:13:2: undeclared name: abort
C:\Go\src\runtime\panic.go:20:2: undeclared name: abort
C:\Go\src\runtime\print.go:14:3: undeclared name: putchar
```

I have no idea what's going on here. Shouldn't C:\Go contain the regular Go installation instead of TinyGo? TinyGo requires there to be a regular Go installation separate from the TinyGo installation.

hbroer, 2020-09-05:
Yes, `C:\Go` contains a regular Go installation. TinyGo is in `C:\tinygo`.

aykevl, 2020-09-05:
No, it appears that you have files from TinyGo in C:\Go (the panic.go error message can only come from a TinyGo runtime file). Perhaps you can try removing and reinstalling Go to fix this?

> Edit: I think I cant use `go test` because I use `machine.Pin` etc.

No that would be difficult.

hbroer, 2020-09-05:
hm I don't think so:

`C:\Go\src\runtime\panic.go`
```
// Copyright 2014 The Go Authors. All rights reserved.
// Use of this source code is governed by a BSD-style
// license that can be found in the LICENSE file.

package runtime

import (
	"runtime/internal/atomic"
	"runtime/internal/sys"
	"unsafe"
)

// We have two different ways of doing defers. The older way involves creating a

....and so on...
```

My go app also run fine. Just tested it out.

deadprogram, 2024-11-30:
Closing since it appears question has been answered. Thanks everyone!