tinygo-org/tinygo #1360: Can we add a go.mod file to the packages?

Nerzal, 2020-09-05:
Hi, 

it would greatly help setting up code completion and (gopls in general) to work with TinyGo, when the device, machine, os, reflect, runtime, sync, and syscall packages would have a go.mod file.

Example go.mod file i use in machine package to get code completion: 

```go
module github.com/tinygo-org/tinygo/src/machine

go 1.14
```

related PR on tinygo-site: https://github.com/tinygo-org/tinygo-site/pull/107

aykevl, 2020-09-07:
I have found a way that is much simpler and should work better. It does not require go.mod files. I will make a PR to update the documentation.

aykevl, 2020-09-07:
Here it is: https://github.com/tinygo-org/tinygo-site/pull/112

hbroer, 2020-09-07:
Yea I tried to make the project a go module, but had a problem with intellisense. I forgot to post this yesterday, so here it is:

It does not work good with modules enabled. I tried to add just one driver and then I had to add a bunch of go.mod files and that will break code inspection when jumping trough the sources inside of the drivers:

```
module my/project

go 1.14

replace machine => C:\tinygo\src\machine
replace runtime => C:\tinygo\src\runtime
replace device => C:\tinygo\src\device
replace device/arm => C:\tinygo\src\device\arm
replace device/avr => C:\tinygo\src\device\avr
replace device/kendryte => C:\tinygo\src\device\kendryte
replace device/nrf => C:\tinygo\src\device\nrf
replace device/nxp => C:\tinygo\src\device\nxp
replace device/riscv => C:\tinygo\src\device\riscv
replace device/sam => C:\tinygo\src\device\sam
replace device/sifive => C:\tinygo\src\device\sifive
replace device/stm32 => C:\tinygo\src\device\stm32

require (
	device v0.0.0-00010101000000-000000000000 // indirect
	device/arm v0.0.0-00010101000000-000000000000 // indirect
	device/avr v0.0.0-00010101000000-000000000000 // indirect
	device/kendryte v0.0.0-00010101000000-000000000000 // indirect
	device/nrf v0.0.0-00010101000000-000000000000 // indirect
	device/nxp v0.0.0-00010101000000-000000000000 // indirect
	device/riscv v0.0.0-00010101000000-000000000000 // indirect
	device/sam v0.0.0-00010101000000-000000000000 // indirect
	device/sifive v0.0.0-00010101000000-000000000000 // indirect
	device/stm32 v0.0.0-00010101000000-000000000000 // indirect
	machine v0.0.0-00010101000000-000000000000
	runtime v0.0.0-00010101000000-000000000000 // indirect
	tinygo.org/x/drivers v0.13.0
)
```

Nerzal, 2020-09-15:
So i'd say we can close this issue ? :D

aykevl, 2020-09-15:
I think this issue has been addressed. While there are no go.mod files (yet), IDE support has much improved with better documentation and a working VS Code extension: https://github.com/tinygo-org/vscode-tinygo

closed 2020-09-15 by aykevl