tinygo-org/tinygo #1364: inconsistency of reflect.SliceHeader's field types with official reflect package

mathetake, 2020-09-07:
When I was working with reflect.SliceHeader of TinyGo, I found that the type of `SliceHeader.{Len,Cap}` is `uintptr` and is different from the one of Go's official implementation which is `int`:

- tinygo: https://github.com/tinygo-org/tinygo/blob/720a54a0fec0720fd7517286810d0837eee2a38a/src/reflect/value.go#L695-L696
- go: https://github.com/golang/go/blob/3dec253783e1211989102ac6abd34cddbf8ba0e6/src/reflect/value.go#L1982-L1983

is this intentional?  If not, may I fix this and replace `uintptr` with `int` ?

motivation: If possible, I do not want to have separated files for Go (for testing) and Tinygo (for building) of virtually identical files
- https://github.com/mathetake/proxy-wasm-go/blob/ff0f378ab727bc314c72ab77bf66111903edf443/runtime/hostcall_utils_go.go#L27-L41
- https://github.com/mathetake/proxy-wasm-go/blob/ff0f378ab727bc314c72ab77bf66111903edf443/runtime/hostcall_utils_tinygo.go#L27-L41

deadprogram, 2020-09-09:
Please see https://github.com/tinygo-org/tinygo/issues/1284 for a discussion about this topic.

mathetake, 2020-09-09:
thanks for the pointer

closed 2020-09-09 by mathetake