tinygo-org/tinygo #1366: tinygo dbg does not exit

sago35, 2020-09-07:
The quit command in gdb does not exit tinygo.
If you kill the openocd process, tinygo will exit.

```
$ ./bin/tinygo-0.15.0-dev-de3ffe0 gdb --target feather-m4 --size short --programmer jlink ./src/examples/b1
   code    data     bss |   flash     ram
   7492      12    6344 |    7504    6356
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /tmp/tinygo778579600/main...
Remote debugging using :3333
(*runtime/volatile.Register8).Get (r=0x2000183c <runtime.timerWakeup>) at /home/sago35/tinygo/tinygo/src/runtime/volatile/register.go:18
18		return LoadUint8(&r.Reg)
Loading section .text, size 0x1d44 lma 0x4000
Loading section .tinygo_stacksizes, size 0x4 lma 0x5d44
Loading section .data, size 0x8 lma 0x5d48
Start address 0x00005542, load size 7504
Transfer rate: 21 KB/sec, 2501 bytes/write.
SWD DPIDR 0x2ba01477
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000058c msp: 0x2000d6a0
(gdb) q
A debugging session is active.

	Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) y
Detaching from program: /tmp/tinygo778579600/main, Remote target
Ending remote debugging.
[Inferior 1 (Remote target) detached]

```

sago35, 2020-09-07:
https://github.com/tinygo-org/tinygo/blob/release/main.go#L400

It appears to be working correctly if I add the following as noted here

```
 			defer func() {
 				daemon.Process.Signal(os.Interrupt)
 				// Maybe we should send a .Kill() after x seconds?
+				daemon.Process.Kill()
 				daemon.Wait()
 			}()
```

aykevl, 2020-09-09:
I tested it but it works for me. What OS are you using?

sago35, 2020-09-09:
I'm using ubuntu 2020.04.
And openocd latest.

```
Open On-Chip Debugger 0.10.0+dev-01404-g393448342 (2020-09-10-08:23)
```

kenbell, 2020-12-08:
I see the same on MacOS Catalina (10.15.7):
```
tinygo version 0.16.0 darwin/amd64 (using go version go1.14.6 and LLVM version 11.0.0)
Open On-Chip Debugger 0.10.0+dev-01517-g7b641d3d4-dirty (2020-12-05-12:30)
```

sago35, 2021-02-26:
same on Windows 10.

```
$ tinygo version
tinygo version 0.17.0-dev-801bd2a windows/amd64 (using go version go1.15.6 and LLVM version 11.0.0)
```

sago35, 2021-02-26:
Does it depend on the version of openocd?
What version of openocd do you have? @aykevl

aykevl, 2021-02-27:
It has always worked well for me, with `tinygo gdb -target=microbit examples/serial` for example. I have OpenOCD version 0.10.0 and am using `gdb-multiarch` version 8.2.1.

I normally press Ctrl-D to exit but using `q` or `quit` also works for me.

kenbell, 2021-02-27:
This is what i see:
[image]

aykevl, 2021-02-27:
Weird.
I don't think killing the process right away is the appropriate solution. Perhaps killing it after a few milliseconds would be better? For example:

```go
daemon.Process.Signal(os.Interrupt)\
go func() {
    time.Sleep(time.Millisecond * 100)
    daemon.Process.Kill()
}()
daemon.Wait()
```

I cannot reproduce it locally so I'm not sure how to test such a fix.

sago35, 2021-02-28:
@aykevl 
It worked well.
I'll create a pull request later.

deadprogram, 2021-04-22:
This was included in the last release, so now closing. Please reopen if this is in error. Thanks!

PR #1684 (2021-03-05): gdb: kill openocd if it does not exit
Fixes #1366
closed 2021-04-22 by deadprogram