tinygo-org/tinygo #1370: Tinygo + gopls doesn't work.

adamijak, 2020-09-09:
`package main

import "machine"
import "time"

func main() {
    led := machine.Led
    led.Configure(machine.PinConfig{Mode: machine.PinOutput})
    for {
        led.Low()
        time.Sleep(time.Millisecond * 300)

        led.High()
        time.Sleep(time.Millisecond * 1000)
    }
}`
Setting `GOPATH="/path/to/go:/usr/lib/tinygo/src"` will allows me to include machine. 
But still I get error `Led not declared in package machine`. That's caused by first line of board_bluepill.go `// +build bluepill` so its not detected as package machine for gopls.

Any syggestions how to fix this to get gopls working with tiny go.

aykevl, 2020-09-09:
This is something we're working on, see: https://github.com/tinygo-org/tinygo-site/pull/112

adamijak, 2020-09-09:
Maybe we have found solution on this thread. Check this out https://github.com/golang/go/issues/41290

aykevl, 2020-09-09:
Yes, this is what https://github.com/tinygo-org/tinygo-site/pull/112 also does (and more).

Closing this now as the issue has been resolved.

closed 2020-09-09 by aykevl