tinygo-org/tinygo #1374: Is it possible to create aligned RAM dynamically?

sago35, 2020-09-12:
I am making a DMAC driver for samd51.
I want to create 128-bit aligned RAM dynamically.

Now I'm writing the following, does the following work correctly?
So far, it appears to be located at an address that is 128-bit aligned.

```go
func NewDescriptor(cfg DescriptorConfig) *DMADescriptor {                                                                      
    //go:align 16
    var ret DMADescriptor
    ret.UpdateDescriptor(cfg)
    return &ret
}
```

I know that the Descriptor specified in BASEADDR can be specified as follows
I want to dynamically create the next Descriptor after the Descriptor specified in BASEADDR.

https://github.com/tinygo-org/things/blob/master/hub75/driver_samd51.go#L15

aykevl, 2020-09-12:
>  So far, it appears to be located at an address that is 128-bit aligned.

That will be purely by chance. The `//go:align` pragma is only respected on global variables.

One way to do it is to allocate a larger slice and align it manually. Something like this (untested):

```go
slice := make([]byte, length+15)
ptr := uintptr(unsafe.Pointer(&slice[0]))
slice = slice[15-(ptr+15)%16:]
```

sago35, 2020-09-12:
Thank you for your reply.
I created the following.
It's probably working fine.

```
package align

import (
	"unsafe"
)

func Make(alignBytes, length int) []byte {
	slice := make([]byte, length+alignBytes-1)
	slice = Align(slice, alignBytes)
	return slice[:length]
}

func Align(in []byte, alignBytes int) []byte {
	ptr := uintptr(unsafe.Pointer(&in[0]))
	slice := in[(alignBytes-1)-((int(ptr)+alignBytes-1)%alignBytes):]
	return slice
}
```

```
=== RUN   TestAlign16
    TestAlign16: align_test.go:34: align 0: input 006634E0, got 006634E0
    TestAlign16: align_test.go:34: align 1: input 006634E1, got 006634F0
    TestAlign16: align_test.go:34: align 2: input 006634E2, got 006634F0
    TestAlign16: align_test.go:34: align 3: input 006634E3, got 006634F0
    TestAlign16: align_test.go:34: align 4: input 006634E4, got 006634F0
    TestAlign16: align_test.go:34: align 5: input 006634E5, got 006634F0
    TestAlign16: align_test.go:34: align 6: input 006634E6, got 006634F0
    TestAlign16: align_test.go:34: align 7: input 006634E7, got 006634F0
    TestAlign16: align_test.go:34: align 8: input 006634E8, got 006634F0
    TestAlign16: align_test.go:34: align 9: input 006634E9, got 006634F0
    TestAlign16: align_test.go:34: align 10: input 006634EA, got 006634F0
    TestAlign16: align_test.go:34: align 11: input 006634EB, got 006634F0
    TestAlign16: align_test.go:34: align 12: input 006634EC, got 006634F0
    TestAlign16: align_test.go:34: align 13: input 006634ED, got 006634F0
    TestAlign16: align_test.go:34: align 14: input 006634EE, got 006634F0
    TestAlign16: align_test.go:34: align 15: input 006634EF, got 006634F0
    TestAlign16: align_test.go:34: align 16: input 006634F0, got 006634F0
    TestAlign16: align_test.go:34: align 17: input 006634F1, got 00663500
```

closed 2020-09-12 by sago35