tinygo-org/tinygo #1380: ATtiny13A

Racso201, 2020-09-13:
Hi,

Would it be possible to write TinyGo code for a ATtiny13A based on the AVR Architecture?

While the ATtiny13A hardware is very limited, it could create some exciting projects and proof of concepts for TinyGo running on a device this small.

I have been following the TinyGo project for years and am not sure if this is the type of Micrcontroller that TinyGo will one day be able to run from?

Just incase it helps, the ATtiny13A only has 1K bytes of flash and 64 bytes of SRAM.

aykevl, 2020-09-13:
Theoretically, maybe. In practice, I don't really see it as a viable target for TinyGo. Certainly you wouldn't be able to use a GC or goroutines, so most of the benefits of Go are already gone. If it ever gets any support, it would only be as a proof of concept. The lowest level chip that currently kind of works is the ATtiny85 as used in the DigiSpark (8kB flash, 512 bytes RAM).

Racso201, 2020-09-13:
Understood, thank you :)

closed 2020-09-13 by Racso201