tinygo-org/tinygo #1381: ESP-NOW

Racso201, 2020-09-13:
Hi,

Are there any plans to implement the ESP-NOW protocol in a future version of TinyGo. I understand minimal board support will possibly be included in 0.15?

This is a amazing protocol that allows communication between boards.

Thank You

aykevl, 2020-09-13:
No, not in the near future. And most likely not in TinyGo itself but in a separate package. ESP-NOW is (as far as I know) a proprietary protocol, and personally I'm more interested in standard protocols like WiFi, Bluetooth and perhaps other (mesh?) protocols.

To get an idea of what support exists right now: you can use all available GPIO pins and I have a PR ready that adds SPI support (#1379). No other peripherals (I2C, I2S, PWM, ADC, etc) are supported and there is certainly no wireless support right now.

Racso201, 2020-09-13:
Thank you for getting back to me,

Very impressed by the progress with ESP and TinyGo.

I certainly would like to see this protocol as a separate package in the future. ESP-NOW allows longer distance transmission and lower power consumption without having to connect to the WIFI connection each time the device turns on.

Thank you for the work you have put into TinyGo :)

closed 2020-09-13 by Racso201