tinygo-org/tinygo #1395: (ESP8266/ESP32) undefined reference to `__divsf3' and `__fixdfdi'

alankrantas, 2020-09-18:
When I tried to run some of my script on ESP8266/ESP32, I got the following message:

```
C:\Users\Admin\AppData\Local\Temp\tinygo386861051\main.o:(.literal.main.main+0x24): undefined reference to `__divsf3'
error: failed to link C:\Users\Admin\AppData\Local\Temp\tinygo386861051\main: exit status 1
```

It seems to have something to do with dividing float32 numbers (change them to float64 might solve the problem), but I haven't been able to replicate this in a new script. 

I also get a "undefined reference to `__fixdfdi'" message in some scripts. Seems to be related with convert floating numbers to int64.

aykevl, 2020-09-18:
Yes, this is a known issue. Normally we would use compiler-rt to fix this, but that is not currently possible because of a bug in the Xtensa fork of LLVM: https://github.com/espressif/llvm-project/issues/33. So until that bug is fixed, we can't do much sadly.

jbpratt, 2020-11-23:
Seems this can be closed, the fix was supposedly merged in here https://github.com/espressif/llvm-project/commit/348e54d4164b32c862dc06069a44a01d85903b52

aykevl, 2020-12-01:
It is not yet fixed, this also needs some changes in TinyGo (but those changes were blocked on that bugfix in LLVM). Hopefully soon.

aykevl, 2020-12-03:
For the ESP32, it will be fixed with https://github.com/tinygo-org/tinygo/pull/1522. The ESP8266 needs a bit more work (it needs to set the `-mcpu` flag while compiling the compiler-rt library) and will hopefully follow soon after that one gets merged.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram