tinygo-org/tinygo #1396: (ESP8266) Pin.Get() undefined

alankrantas, 2020-09-18:
For ESP8266 type machine.Pin has no field or method Get.

Also, should ESP8266 and ESP32 define machine.LED (Pin(2) on most boards) as well?

aykevl, 2020-09-18:
> For ESP8266 type machine.Pin has no field or method Get.

Yes, this is not yet implemented. Feel free to submit a patch :)

> Also, should ESP8266 and ESP32 define machine.LED (Pin(2) on most boards) as well?

`machine.LED` is defined for the esp32-wroom-32 as it has an on-board LED. It is not defined for the esp8266 target as it is not guaranteed to have this LED (even though almost all development boards have it). So if we add support for a board (and not just the chip itself), we can define `machine.LED`.

jbpratt, 2020-11-23:
I think this can be closed!

alankrantas, 2020-11-23:
Just tested it, works in both PinInput and PinOutput mode. Thanks everyone!

closed 2020-11-23 by alankrantas