tinygo-org/tinygo #1400: support debug.ReadGCStats function: implement reflect.Swapper

mathetake, 2020-09-20:
I would like to have runtime GC stats for performance analysis, so I tried to implement here: https://github.com/mathetake/tinygo/commit/f98ded439ed69cd0acf84682365bf5a7aa1ddc1e

What I found was debug.ReadGCStats [uses sort.Slice](https://github.com/golang/go/blob/go1.15.2/src/runtime/debug/garbage.go#L74), and it [uses reflect.Swapper](https://github.com/golang/go/blob/go1.15.2/src/sort/slice.go#L15) which is [yet to be implemented](https://github.com/tinygo-org/tinygo/blob/v0.15.0/src/reflect/swapper.go#L4) in TinyGo.

I think there are two different options to solve this issue:

1. implement debug package in TinyGo without using sort.Slice
2. implement reflect.Swapper

I would love to do implementation but not sure which way we should take. 

Any thoughts? Thanks!
___

P.S. I found that the TinyGo's website says that `sort` package is supported but this is wrong, isn't is?  https://github.com/tinygo-org/tinygo-site/blob/release/content/lang-support/stdlib.md#title-packages-supported-by-tinygo

aykevl, 2020-09-20:
What exactly do you need from `ReadGCStats`? Adding full support for it would likely be a significant cost even when it is not used (in binary size, memory consumption and runtime cost) while some subset might be near-zero cost.

> P.S. I found that the TinyGo's website says that `sort` package is supported but this is wrong, isn't it? https://github.com/tinygo-org/tinygo-site/blob/release/content/lang-support/stdlib.md#title-packages-supported-by-tinygo

Please read the top text. It only says you can import the `sort` package, it does not say it will work.

mathetake, 2020-09-20:
I just wanted to compare and analyze different GC algorithms (custom implementations and built ones) with `ReadGCStats`, but yes It will increase the code size if we enable by default.

Realistically, it is enough to do benchmarks in more generic ways (like measuring request per seconds) without using `ReadGCStats`. I just thought it would be helpful to develop GC algorithms.

> it does not say it will work.

So sorry I didn't read the top text. got it

mathetake, 2020-09-20:
https://github.com/golang/go/blob/master/src/runtime/debug/garbage.go#L16-L17

I think these two fields are enough to get general idea of GC performance

aykevl, 2020-09-22:
If this is for use during TinyGo development, it might be possible to include this but disable it by default, similar to how there is some logging code included but disabled by default for the scheduler:

https://github.com/tinygo-org/tinygo/blob/1096596b692bd1cbec9b58336f31313d9faaeb0f/src/runtime/scheduler.go#L21

Because this is a `const`, any `if` statement with it will be trivial to optimize away by the compiler and should have no effect on the final binary.

davecheney, 2026-10-26:
Thanks for the proposal. This still seems useful for GC performance analysis.

`runtime.ReadMemStats` now exposes `NumGC`, but `debug.ReadGCStats` remains a no-op. Supporting the collection count should be straightforward. Pause statistics need more care to measure actual stop-the-world time rather than the entire collection, and to avoid unnecessary overhead on small devices.