tinygo-org/tinygo #1401: [Support] Arduino Pro Mini, ATmega328P, 5V/16MHz

ProximaB, 2020-09-21:
I have manage to compile tinygo and upload it to Arduino Mini Pro with avrisp mkii and prove that it works.
I used avrdude tool to upload it with serial programmer.
```bash
sudo tinygo build -target arduino -o output.hex main.go
avrdude -avrdude.conf -v -patmega328p -carduino -P/dev/ttyUSB0 -b57600 -D -Uflash:w:./output.hex 
 ```
Config used (basic arduino avrdude config): https://pastebin.com/xVtzAfm4

```
avrdude: verifying flash memory against ./output.hex:
avrdude: load data flash data from input file ./output.hex:
avrdude: input file ./output.hex auto detected as Intel Hex
avrdude: input file ./output.hex contains 558 bytes
```

aykevl, 2020-09-22:
Cool!
Is this a request to add support for this board?

ProximaB, 2020-09-22:
Yeah, that's it. We could add some instructions on the website: https://tinygo.org/microcontrollers/. That would be great.

aykevl, 2020-09-22:
Actually, it would be best to add native support for this in TinyGo. Seeing this, it should not be very difficult. It looks like the board is very similar to the Arduino Uno.

ProximaB, 2020-09-22:
Alright, I would like to contribute and add native support for Arduino Pro mini to TinyGo.

duarte2025, 2020-09-25:
For arduino pro mini would not be the pin mapping? Well, the microcontroller is the same as the arduino one, if I'm not mistaken.

ProximaB, 2020-09-25:
@duarte2025 That's right. I would like to add some instructions how to boot devices as well, because I had some troubles with avrdude.

ProximaB, 2020-09-25:
I have noticed that the code lack of consistence in naming analog pins or one way of structuring code. That will cause code portability issues. 
[board_arduino.go#L12](https://github.com/ProximaB/tinygo/blob/6b8940421e169a751dd039d8c7d1b1c4a1465fb2/src/machine/board_arduino.go#L12
)
```go
	D0  = PD0 // RX
	D1  = PD1 // TX
...
	ADC0 Pin = PC0
	ADC1 Pin = PC1
....
	UART_TX_PIN Pin = PD1
	UART_RX_PIN Pin = PD0
```
[board_arduino_nano33.go#L35](https://github.com/ProximaB/tinygo/blob/d1c4ed664e540e00b3d18ff340d6798c45b382cf/src/machine/board_arduino_nano33.go#L35
)
```go
	RX0 Pin = PB23 // UART2 RX
	TX1 Pin = PB22 // UART2 TX
...
	A0 Pin = PA02 // ADC/AIN[0]
	A1 Pin = PB02 // ADC/AIN[10]
```

deadprogram, 2024-12-13:
Seems like this was already taken care of a while ago. Now closing, please reopen if needed. Thanks!

closed 2024-12-13 by deadprogram