tinygo-org/tinygo #1408: findPinPadMapping panics (index out of range) when noPin is passed

conejoninja, 2020-09-25:
On the *pybadge* it's not possible to use the onboard display, the error might be caused by this change : https://github.com/tinygo-org/tinygo/commit/1451eeaf414176f2b96298516f3bf2c3296f8037

SPI1_SDI_PIN is defined as noPin (255) https://github.com/tinygo-org/tinygo/blob/release/src/machine/board_pybadge.go#L103
pinPadMapping is 64 length https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_atsamd51.go#L267

Code to reproduce the issue

```go
	machine.SPI1.Configure(machine.SPIConfig{
		SCK:       machine.SPI1_SCK_PIN,
		SDO:       machine.SPI1_SDO_PIN,
		SDI:       machine.SPI1_SDI_PIN,
		Frequency: 8000000,
	})
```

sago35, 2020-11-03:
@conejoninja 
You can maybe set up a PB12 instead of NoPin as with the arduino.
You can't read the value because it's not connected, but I think the SPI setup will work!

https://github.com/adafruit/ArduinoCore-samd/blob/master/variants/pybadge_m4/variant.cpp#L104

> This fix the issue, but not sure about the implications : [conejoninja@6c39859](https://github.com/conejoninja/tinygo/commit/6c39859ec32b0d28ce3aadd0f0966079fc5cc380)

On a different note, I think it would be good to change it so that it doesn't become panic.
So I think this change is necessary as well.

sago35, 2020-11-04:
https://github.com/adafruit/ArduinoCore-samd/blob/master/variants/pygamer_m4/variant.cpp#L100-L101

PB12 is doubly defined here.
This seems to be a mistake on the Arduino side.

aykevl, 2020-11-12:
@conejoninja your proposed fix looks correct. Can you make a PR with it?

conejoninja, 2020-11-12:
@aykevl similar code was already merged here : https://github.com/tinygo-org/tinygo/commit/db27541b1a44a903feeeef91840314a56fcdc725

we still need to consider adding an extra check in findPinPadMapping so pin/2 isn't out of range

aykevl, 2020-11-16:
Sorry I didn't see that was for samd51, somehow I thought it was for samd21. Yes, for the samd51 it is indeed already fixed. See #1500.

> we still need to consider adding an extra check in findPinPadMapping so pin/2 isn't out of range

Yes, #1491.

aykevl, 2020-12-02:
#1491 has been merged so this bug is fixed.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram