tinygo-org/tinygo #1415: I2C issue in Arduino

sakthimaan, 2020-09-26:
Getting below error when running bmp180 example code. 

**Program Code:** 
``` 
https://raw.githubusercontent.com/tinygo-org/drivers/release/examples/bmp180/main.go
```

**Command Executed:**
```bash
 tinygo build -o test.hex -target arduino bmp180.go
 ```
**Tinygo Version:** 
``` tinygo version 0.15.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1) ```

**tinygo env**
```
GOOS="linux"
GOARCH="amd64"
GOROOT="/home/***/sdk/go1.15.2"
GOPATH="/home/***/go:/home/ram/tinygo:"
GOCACHE="/home/***/.cache/tinygo"
CGO_ENABLED="1"
TINYGOROOT="/usr/local/lib/tinygo"
```

**Error:**
```
tinygo build -o test.hex -target arduino bmp180.go 
/tmp/tinygo299647481/main.o: In function `runtime.printfloat32':
bmp180.go:(.text.runtime.printfloat32+0x3a): undefined reference to `__extendsfdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x74): undefined reference to `__adddf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found address size '9', this reader can not handle sizes greater than '8'.
bmp180.go:(.text.runtime.printfloat32+0xc2): undefined reference to `__eqdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '12', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0xec): undefined reference to `__unorddf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x140): undefined reference to `__eqdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x16a): undefined reference to `__unorddf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '4866', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x2aa): undefined reference to `__subdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '13312', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x2ca): undefined reference to `__divdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '256', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x2d6): undefined reference to `__gedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '65535', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x316): undefined reference to `__ltdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '19', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x330): undefined reference to `__divdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '15619', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x368): undefined reference to `__gedf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x386): undefined reference to `__muldf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x3c2): undefined reference to `__adddf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x3dc): undefined reference to `__ltdf2'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '1537', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x3fa): undefined reference to `__divdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x46e): undefined reference to `__fixdfsi'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x484): undefined reference to `__floatsidf'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '1792', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x49e): undefined reference to `__subdf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '0', this reader only handles version 2, 3 and 4 information.
bmp180.go:(.text.runtime.printfloat32+0x4b8): undefined reference to `__muldf3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '19458', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo299647481/main.o: In function `LBB19_3':
bmp180.go:(.text.command-line-arguments.main+0x53e): undefined reference to `__lshrsi3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '13312', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo299647481/main.o: In function `LBB19_38':
bmp180.go:(.text.command-line-arguments.main+0xce4): undefined reference to `__ashlsi3'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: Dwarf Error: found dwarf version '256', this reader only handles version 2, 3 and 4 information.
/tmp/tinygo299647481/main.o: In function `LBB19_44':
bmp180.go:(.text.command-line-arguments.main+0xdc8): undefined reference to `__lshrsi3'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo299647481/main: exit status 1
```

aykevl, 2020-09-27:
This is a known issue. Most of these errors are caused by the use of `float64`, which is not supported by avr-libc. Hopefully this can be fixed after we switch to LLVM 11.

You could try this instead, which avoids floating point numbers:

```go
package main

import (
	"time"

	"machine"

	"tinygo.org/x/drivers/bmp180"
)

func main() {
	machine.I2C0.Configure(machine.I2CConfig{})
	sensor := bmp180.New(machine.I2C0)
	sensor.Configure()

	connected := sensor.Connected()
	if !connected {
		println("BMP180 not detected")
		return
	}
	println("BMP180 detected")

	for {
		temp, _ := sensor.ReadTemperature()
		println("Temperature:", temp, "°C/1000")

		pressure, _ := sensor.ReadPressure()
		println("Pressure", pressure, "mPa")

		time.Sleep(2 * time.Second)
	}
}
```

sakthimaan, 2020-09-27:
> This is a known issue. Most of these errors are caused by the use of `float64`, which is not supported by avr-libc. Hopefully this can be fixed after we switch to LLVM 11.
>
> You could try this [...]

**Still facing the Error**
```
>tinygo build -o test.hex -target arduino bmp180.go
c:/users/rsubbura/scoop/apps/avr-gcc/current/bin/../lib/gcc/avr/10.1.0/../../../../avr/bin/ld.exe: C:\Users\rsubbura\AppData\Local\Temp\tinygo182447759\main.o: in function `LBB20_3':
bmp180.go:(.text.command-line-arguments.main+0x532): undefined reference to `__lshrsi3'
c:/users/rsubbura/scoop/apps/avr-gcc/current/bin/../lib/gcc/avr/10.1.0/../../../../avr/bin/ld.exe: C:\Users\rsubbura\AppData\Local\Temp\tinygo182447759\main.o: in function `LBB20_38':
bmp180.go:(.text.command-line-arguments.main+0xcba): undefined reference to `__ashlsi3'
c:/users/rsubbura/scoop/apps/avr-gcc/current/bin/../lib/gcc/avr/10.1.0/../../../../avr/bin/ld.exe: C:\Users\rsubbura\AppData\Local\Temp\tinygo182447759\main.o: in function `LBB20_44':
bmp180.go:(.text.command-line-arguments.main+0xd9c): undefined reference to `__lshrsi3'
collect2.exe: error: ld returned 1 exit status
error: failed to link C:\Users\rsubbura\AppData\Local\Temp\tinygo182447759\main: exit status 1

```

aykevl, 2020-09-27:
Yes I know. As I said in the pull request, it does not completely solve this issue.

deadprogram, 2021-03-24:
Just wondering if this has now been addressed in the latest `dev` branch?

aykevl, 2021-03-24:
No. The chance of it happening is slightly reduced (when printing error types) but the core issue is not fixed.

sakthimaan, 2021-06-29:
is this issue fixed?

PR #1416 (2020-10-02): runtime: use dedicated printfloat32
It can be unexpected that printing a float32 involves 64-bit floating
point routines, see for example:
https://github.com/tinygo-org/tinygo/issues/1415

This commit adds a dedicated printfloat32 instead just for printing
float32 values. It comes with a possible code size increase, but only if
both float32 and float64 values are printed. Therefore, this should be
an improvement in almost all cases.

I also tried using printfloat32 for everything (and casting a float64 to
float32 to print) but the printed values are slightly different,
breaking the testdata/ma [... cut]