tinygo-org/tinygo #1418: wasm tests timeout on i686

QuLogic, 2020-09-28:
While updated to 0.15.0, I also started running the wasm tests. This works on 64-bit systems, but fails on 32-bit: https://koji.fedoraproject.org/koji/taskinfo?taskID=52392695
```
+ make wasmtest
go test ./tests/wasm
2020/09/28 07:13:24 Starting server at ":8826" for dir: /tmp/wasm_test630700884
2020/09/28 07:13:24 Command: tinygo build -o /tmp/wasm_test630700884/chan.wasm -target wasm testdata/chan.go; err=<nil>; full output:
2020/09/28 07:13:25 Command: tinygo build -o /tmp/wasm_test630700884/event.wasm -target wasm testdata/event.go; err=<nil>; full output:
2020/09/28 07:13:28 Command: tinygo build -o /tmp/wasm_test630700884/fmt.wasm -target wasm testdata/fmt.go; err=<nil>; full output:
2020/09/28 07:13:29 Command: tinygo build -o /tmp/wasm_test630700884/fmtprint.wasm -target wasm testdata/fmtprint.go; err=<nil>; full output:
--- FAIL: TestChan (5.70s)
    chan_test.go:31: unable to execute *log.EnableParams: context deadline exceeded
2020/09/28 07:13:30 Command: tinygo build -o /tmp/wasm_test630700884/log.wasm -target wasm testdata/log.go; err=<nil>; full output:
--- FAIL: TestEvent (6.26s)
    event_test.go:41: log1: 
    event_test.go:42: log2: 
    event_test.go:44: context deadline exceeded
--- FAIL: TestFmt (8.96s)
    fmt_test.go:38: log1: 
    fmt_test.go:40: unable to execute *log.EnableParams: context deadline exceeded
--- FAIL: TestFmtprint (10.36s)
    fmtprint_test.go:34: log1: 
    fmtprint_test.go:36: unable to execute *log.EnableParams: context deadline exceeded
--- FAIL: TestLog (11.01s)
    log_test.go:39: log1: 
    log_test.go:41: unable to execute *log.EnableParams: context deadline exceeded
FAIL
FAIL	_/builddir/build/BUILD/tinygo-0.15.0/tests/wasm	11.010s
FAIL
```
I tried increasing the timeouts from 5 to 15 seconds, but that didn't seem to have an effect. I'm not sure if this is a bug in tinygo, or the chrome driver stuff.

aykevl, 2020-09-28:
I'll try reproducing this in an i386 Debian container.

aykevl, 2020-09-28:
I doubt this is a bug in the compiler itself, as these tests are run in CI and succeed there (and wasm output should be independent of the host). But maybe a bug in the test runner, or in the Chrome driver as you mentioned.

aykevl, 2020-09-29:
I can reproduce this:

```
root@buster-i386:~/tinygo# make wasmtest
go test ./tests/wasm
2020/09/29 16:56:09 Starting server at ":8826" for dir: /tmp/wasm_test050927264
2020/09/29 16:56:13 Command: tinygo build -o /tmp/wasm_test050927264/chan.wasm -target wasm testdata/chan.go; err=<nil>; full output:
--- FAIL: TestChan (8.90s)
    chan_test.go:31: context deadline exceeded
2020/09/29 16:56:18 Command: tinygo build -o /tmp/wasm_test050927264/fmt.wasm -target wasm testdata/fmt.go; err=<nil>; full output:
2020/09/29 16:56:19 Command: tinygo build -o /tmp/wasm_test050927264/fmtprint.wasm -target wasm testdata/fmtprint.go; err=<nil>; full output:
2020/09/29 16:56:21 Command: tinygo build -o /tmp/wasm_test050927264/log.wasm -target wasm testdata/log.go; err=<nil>; full output:
2020/09/29 16:56:21 Command: tinygo build -o /tmp/wasm_test050927264/event.wasm -target wasm testdata/event.go; err=<nil>; full output:
--- FAIL: TestFmt (14.69s)
    fmt_test.go:38: log1: 
    fmt_test.go:40: context deadline exceeded
--- FAIL: TestFmtprint (15.42s)
    fmtprint_test.go:34: log1: 
    fmtprint_test.go:36: context deadline exceeded
--- FAIL: TestEvent (8.56s)
    event_test.go:41: log1: <span>1<br></span><span>4<br></span>
    event_test.go:42: log2: 
    event_test.go:44: context deadline exceeded
FAIL
FAIL	github.com/tinygo-org/tinygo/tests/wasm	17.494s
FAIL
make: *** [Makefile:380: wasmtest] Error 1
```

Running in a Linux Container (lxc):
OS: Debian buster
Arch: i386 / i686
Chrome: `apt-get install chromium`

Using Debian here simply because I'm more familiar with it, and because it helps to narrow down the issue (it's apparently not a Fedora issue).

aykevl, 2020-09-29:
Looks like a parallelism issue of some sort:

```
root@buster-i386:~/tinygo# go test -v -parallel=1 ./tests/wasm
2020/09/29 17:03:14 Starting server at ":8826" for dir: /tmp/wasm_test490532832
=== RUN   TestChan
=== PAUSE TestChan
=== RUN   TestEvent
=== PAUSE TestEvent
=== RUN   TestFmt
=== PAUSE TestFmt
=== RUN   TestFmtprint
=== PAUSE TestFmtprint
=== RUN   TestLog
=== PAUSE TestLog
=== CONT  TestChan
2020/09/29 17:03:14 Command: tinygo build -o /tmp/wasm_test490532832/chan.wasm -target wasm testdata/chan.go; err=<nil>; full output:
--- PASS: TestChan (1.80s)
=== CONT  TestLog
2020/09/29 17:03:22 Command: tinygo build -o /tmp/wasm_test490532832/log.wasm -target wasm testdata/log.go; err=<nil>; full output:
    TestLog: log_test.go:39: log1: <span>2020/09/29 17:03:23 log 1<br></span><span>2020/09/29 17:03:23 log 2<br></span><span>2020/09/29 17:03:23 log 3<br></span><span>println 4<br></span><span>fmt.Println 5<br></span><span>2020/09/29 17:03:23 log 6<br></span><span>in func 1<br></span><span>2020/09/29 17:03:23 in func 2<br></span>
--- PASS: TestLog (8.08s)
=== CONT  TestFmt
2020/09/29 17:03:27 Command: tinygo build -o /tmp/wasm_test490532832/fmt.wasm -target wasm testdata/fmt.go; err=<nil>; full output:
    TestFmt: fmt_test.go:38: log1: <span>did not panic<br></span>
--- PASS: TestFmt (6.16s)
=== CONT  TestFmtprint
2020/09/29 17:03:35 Command: tinygo build -o /tmp/wasm_test490532832/fmtprint.wasm -target wasm testdata/fmtprint.go; err=<nil>; full output:
    TestFmtprint: fmtprint_test.go:34: log1: <span>test from fmtprint 1<br></span><span>test from fmtprint 2<br></span><span>test from fmtprint 3<br></span><span>test from fmtprint 4<br></span>
--- PASS: TestFmtprint (7.51s)
=== CONT  TestEvent
2020/09/29 17:03:38 Command: tinygo build -o /tmp/wasm_test490532832/event.wasm -target wasm testdata/event.go; err=<nil>; full output:
    TestEvent: event_test.go:41: log1: <span>1<br></span><span>4<br></span>
    TestEvent: event_test.go:42: log2: <span>1<br></span><span>4<br></span><span>2<br></span><span>3<br></span><span>true<br></span>
--- PASS: TestEvent (3.79s)
PASS
ok  	github.com/tinygo-org/tinygo/tests/wasm	27.342s
```

I will continue debugging another time.

aykevl, 2020-09-29:
There seem to be bugs (running on amd64 as the `-race` flag is not available on i386):

```
$ go test -v -race ./tests/wasm
2020/09/29 19:05:03 Starting server at ":8826" for dir: /tmp/wasm_test156534122
=== RUN   TestChan
=== PAUSE TestChan
=== RUN   TestEvent
=== PAUSE TestEvent
=== RUN   TestFmt
=== PAUSE TestFmt
=== RUN   TestFmtprint
=== PAUSE TestFmtprint
=== RUN   TestLog
=== PAUSE TestLog
=== CONT  TestChan
=== CONT  TestFmt
=== CONT  TestLog
=== CONT  TestEvent
2020/09/29 19:05:06 Command: tinygo build -o /tmp/wasm_test156534122/chan.wasm -target wasm testdata/chan.go; err=<nil>; full output:
2020/09/29 19:05:07 Command: tinygo build -o /tmp/wasm_test156534122/event.wasm -target wasm testdata/event.go; err=<nil>; full output:
2020/09/29 19:05:10 Command: tinygo build -o /tmp/wasm_test156534122/fmt.wasm -target wasm testdata/fmt.go; err=<nil>; full output:
fatal error: checkptr: unsafe pointer conversion

goroutine 6 [running]:
runtime.throw(0xc09334, 0x23)
	/usr/local/go/src/runtime/panic.go:1112 +0x72 fp=0xc0000dae18 sp=0xc0000dade8 pc=0x462bb2
runtime.checkptrAlignment(0xc0000e4035, 0xb3a3a0, 0x1)
	/usr/local/go/src/runtime/checkptr.go:13 +0xd0 fp=0xc0000dae48 sp=0xc0000dae18 pc=0x434400
github.com/gobwas/ws.btsEqualFold(0xc0000e4034, 0x9, 0xfcc, 0x10a1730, 0x9, 0x9, 0xc0000e4034)
	/home/ayke/pkg/mod/github.com/gobwas/ws@v1.0.2/util.go:233 +0x184 fp=0xc0000daeb0 sp=0xc0000dae48 pc=0xaaf584
github.com/gobwas/ws.Dialer.Upgrade(0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/home/ayke/pkg/mod/github.com/gobwas/ws@v1.0.2/dialer.go:374 +0xe72 fp=0xc0000db140 sp=0xc0000daeb0 pc=0xaaa752
github.com/gobwas/ws.Dialer.Dial(0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/home/ayke/pkg/mod/github.com/gobwas/ws@v1.0.2/dialer.go:190 +0x444 fp=0xc0000db4c8 sp=0xc0000db140 pc=0xaa8a64
github.com/gobwas/ws.Dial(...)
	/home/ayke/pkg/mod/github.com/gobwas/ws@v1.0.2/dialer.go:47
github.com/chromedp/chromedp.DialContext(0xcd7120, 0xc000098100, 0xc0000ac190, 0x4a, 0xc0000db7e0, 0x1, 0x1, 0x4770e3, 0xc00010da98, 0x2)
	/home/ayke/pkg/mod/github.com/chromedp/chromedp@v0.5.3/conn.go:59 +0xff fp=0xc0000db6a8 sp=0xc0000db4c8 pc=0xae2abf
github.com/chromedp/chromedp.NewBrowser(0xcd71a0, 0xc000208360, 0xc0000ac190, 0x4a, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/ayke/pkg/mod/github.com/chromedp/chromedp@v0.5.3/browser.go:104 +0x3c9 fp=0xc0000db818 sp=0xc0000db6a8 pc=0xadd3c9
github.com/chromedp/chromedp.(*ExecAllocator).Allocate(0xc00022a050, 0xcd71a0, 0xc000208360, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/ayke/pkg/mod/github.com/chromedp/chromedp@v0.5.3/allocate.go:240 +0x1204 fp=0xc0000dbba8 sp=0xc0000db818 pc=0xadb594
github.com/chromedp/chromedp.Run(0xcd71a0, 0xc000208360, 0xc0000dbe30, 0x9, 0x9, 0xc000204120, 0x0)
	/home/ayke/pkg/mod/github.com/chromedp/chromedp@v0.5.3/chromedp.go:227 +0x311 fp=0xc0000dbc98 sp=0xc0000dbba8 pc=0xae04f1
github.com/tinygo-org/tinygo/tests/wasm.TestEvent(0xc000134d80)
	/home/ayke/src/github.com/tinygo-org/tinygo/tests/wasm/event_test.go:25 +0x78a fp=0xc0000dbed0 sp=0xc0000dbc98 pc=0xaf481a
testing.tRunner(0xc000134d80, 0xc198a8)
	/usr/local/go/src/testing/testing.go:992 +0x1ec fp=0xc0000dbfd0 sp=0xc0000dbed0 pc=0x58840c
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc0000dbfd8 sp=0xc0000dbfd0 pc=0x496731
created by testing.(*T).Run
	/usr/local/go/src/testing/testing.go:1043 +0x661
[...]
```

aykevl, 2020-09-29:
This should fix the issue: https://github.com/tinygo-org/tinygo/pull/1420

aykevl, 2020-09-29:
Thank you for reporting this! We're not really testing i386 right now, maybe we should (are there any i386/i686 Go docker images available?).

Somewhat offtopic but I saw that TinyGo is packaged for x86-64, i686, and aarch64. Is there a reason it isn't packaged for ARM (armhf)? Are there any bugs that need to be fixed to make that possible?

QuLogic, 2020-10-01:
> Somewhat offtopic but I saw that TinyGo is packaged for x86-64, i686, and aarch64. Is there a reason it isn't packaged for ARM (armhf)? Are there any bugs that need to be fixed to make that possible?

Well, I tried out a build just now, and the problem seems to be that `go test` segfaults a lot, and there are messages like:
```
/usr/bin/ld: error: /tmp/tinygo752997470/main uses VFP register arguments, /tmp/tinygo752997470/main.o does not
/usr/bin/ld: failed to merge target specific data of file /tmp/tinygo752997470/main.o
collect2: error: ld returned 1 exit status
```

aykevl, 2020-10-01:
Okay, thanks for trying out! It's probably softfloat vs hardfloat and stuff like that, maybe I should try again and let TinyGo produce static executables using musl (I tried this before, but musl isn't easy to compile without running `make`).

QuLogic, 2020-10-01:
Here is a [scratch build](https://koji.fedoraproject.org/koji/taskinfo?taskID=52574275) for armv7hl. For one, I had to skip wasmtest because chromium isn't built there. Then there's the above linkage problem, plus some missing headers. Not entirely sure why that one's happening.

deadprogram, 2020-11-24:
This was released as part of v0.16.0 so now closing. Thanks everyone!

closed 2020-11-24 by deadprogram