tinygo-org/tinygo #1419: SPI Data received is irregularly corrupted

sago35, 2020-09-28:
Creating a WiFi process in WioTerminal.
It repeats the process of receiving 800Byte or so of data, but it corrupts the data about once every few hundred times.
The corruption is random, but non-existent data is received.
At that time, it is not clear if the data is being sent to the SPI bus or not.

```                                                                                                                                                    
NG: <a href="http://rrrrrrrrrginx.com/">nginx.com</a>.</p> 
OK: <a href="http://nginx.com/">nginx.com</a>.</p> 

NG: <h1>Welcome to AAAAAAAnginx!</h1>
OK: <h1>Welcome to nginx!</h1>
```

I had a problem with tinygo's original SPI.Transfer(), but adding more DRE checks did the same thing.
From the log that we have been able to obtain, it appears that RXC is set to 1, even though the reception is not complete.

```go
// tinygo's SPI.Transfer()
func (spi SPI) Transfer(w byte) (byte, error) {
	// write data
	spi.Bus.DATA.Set(uint32(w))

	// wait for receive
	for !spi.Bus.INTFLAG.HasBits(sam.SERCOM_SPIM_INTFLAG_RXC) {
	}

	// return data
	return byte(spi.Bus.DATA.Get()), nil
}
```

```go
// 
func (d *Device) spi_rx(rx []byte) {
	for i := 0; i < len(rx); i++ {
		for !d.bus.Bus.INTFLAG.HasBits(sam.SERCOM_SPIM_INTFLAG_DRE) {
		}
		d.bus.Bus.DATA.Set(uint32(SPT_TAG_DMY))
		for !d.bus.Bus.INTFLAG.HasBits(sam.SERCOM_SPIM_INTFLAG_RXC) {
		}
		rx[i] = byte(d.bus.Bus.DATA.Get())
	}
}
```

ubuntu 20.14
tinygo version 0.16.0-dev-1096596 linux/amd64 (using go version go1.14.4 and LLVM version 10.0.1)

sago35, 2020-09-28:
Addition.

There are some broken data added, like the NG data above, but the SPI slave side seems to know that it is not sending.
This seems to make the communication inconsistent.

sago35, 2020-09-28:
When I added Sleep to try it out, the problem appears to have stopped.

Before adding it: more than 1 in 1000 times
After adding: Not occurred even after 10000 times

```go
func (d *Device) spi_rx(rx []byte) {
	for i := 0; i < len(rx); i++ {
		for !d.bus.Bus.INTFLAG.HasBits(sam.SERCOM_SPIM_INTFLAG_DRE) {
		}
		d.bus.Bus.DATA.Set(uint32(SPT_TAG_DMY))
		for !d.bus.Bus.INTFLAG.HasBits(sam.SERCOM_SPIM_INTFLAG_RXC) {
		}
		rx[i] = byte(d.bus.Bus.DATA.Get())
		time.Sleep(10 * time.Microsecond)
	}
}
```

aykevl, 2020-09-28:
Can you give some more detail on the problem? I don't see a `spi_rx` anywhere.

sago35, 2020-09-29:
I probably shouldn't have written spi_rx() since it was just for example.
Now the problem occurs when I call SPI.Transfer() a few hundred times repeatedly.

Right now, this problem is only happening with SERCOM0 (for WiFi) in WioTerminal.
Writing a similar program with SERCOM0 on the feather-m4 does not cause it to occur.

I hope to be able to write a program to reproduce it, but at the moment I don't know what will happen.

eyz, 2021-12-29:
Is the SAMD51 / SAME51 SPI stack stable now? What are the failure conditions left, if not?

sago35, 2021-12-29:
@eyz 
Thank you for your comment and confirmation.

This problem was caused by a mistake in my settings.
Since there is no problem, #1419 can be closed.

https://github.com/tinygo-org/tinygo/pull/1453#issuecomment-716358412

closed 2021-12-29 by sago35