tinygo-org/tinygo #1422: How to pass wasm arguments to tinyGo function?

thstart, 2020-10-01:
In C I have this function:
`int sum(int a, int b)
{
 return a + b;
}`

Compiling:
`clang -c sum.c`
`emcc -s EXPORTED_FUNCTIONS="['_sum']" sum.c -O3 -o sum.wasm`

Now I can use it like this:

`CREATE TEMP FUNCTION `magic_function`(x ARRAY<INT64>) RETURNS ARRAY<INT64> LANGUAGE js AS '''
const memory = new WebAssembly.Memory({ initial: 256, maximum: 256 });

const env = {
    'abortStackOverflow': _ => { throw new Error('overflow'); },
    'table': new WebAssembly.Table({ initial: 0, maximum: 0, element: 'anyfunc' }),
    'tableBase': 0,
    'memory': memory,
    'memoryBase': 1024,
    'STACKTOP': 0,
    'STACK_MAX': memory.buffer.byteLength,
};

const imports = { env };

const bytes = new Uint8Array([0, 97, 115, 109, 1, 0, 0, 0, 1, 7, 1, 96, 2, 127, 127, 1, 127, 3, 2, 1, 0, 7, 8, 1, 4, 95, 115, 117, 109, 0, 0, 10, 9, 1, 7, 0, 32, 0, 32, 1, 106, 11]);

async function main() {
  const wa = await WebAssembly.instantiate(bytes, imports);
  const exports = wa.instance.exports;
  const magic_sum = exports._sum;
  return x.map((val) => {
    return magic_sum(val, val);
  });
}
return main();

''';
WITH
  numbers AS (
  SELECT
    [1,
    2,
    3] AS y)
SELECT
  y,
  magic_function(y) AS sum
FROM
  numbers; `

How to do the same with the following tinyGo example?
https://wasmbyexample.dev/examples/hello-world/hello-world.go.en-us.html
`package main

// Declare a main function, this is the entrypoint into our go module
// That will be run. In our example, we won't need this
func main() {}

// This exports an add function.
// It takes in two 32-bit integer values
// And returns a 32-bit integer value.
// To make this function callable from JavaScript,
// we need to add the: "export add" comment above the function
//export add
func add(x int, y int) int {
  return x + y;
} `

aykevl, 2020-10-02:
I have updated your question to make it more readable.

I have no idea what you want to do. The blog post you've linked explains how you can call an exported function from JavaScript. As for your problem, I only see a mixture of SQL and JavaScript and I don't know what is supposed to mean. So, what is it that you're trying to do?

thstart, 2020-10-02:
> I have updated your question to make it more readable.
>
> I have no idea what you want to do. The blog post you've linked explains how you can call an exported function from JavaScript. As for your [...]

The question is how to use tinyGo in BigQuery.
BigQuery has the capability to execute custom Javascript functions and wasm.

aykevl, 2020-10-02:
I have never worked with BigQuery so I'm afraid I can't help you with that.

The blog post you linked shows how to build a WebAssembly file. It does have a number of imports however, see targets/wasm_exec.js.

thstart, 2020-10-02:
> I have never worked with BigQuery so I'm afraid I can't help you with that.
> 
> The blog post you linked shows how to build a WebAssembly file. It does have a number of imports however, see targets/wasm_exec.js.

When compiling C version of sum, the symbol is expected as _sum and I can reference it as
const magic_sum = exports._sum;

This is the first question - could I export just a function with tinyGo?

aykevl, 2020-10-03:
> This is the first question - could I export just a function with tinyGo?

Kind of. Again, the linked blog post explains how to export a single function. However, it will also export more than just that so I'm not sure whether it's feasible to use this in BigQuery.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram