tinygo-org/tinygo #1423: The syscall/js.valueInstanceOf not implemented

elct9620, 2020-10-03:
Hi, all

I try to call the `value.InstanceOf(js.Global().Get("String"))` but it seems comment out in the `wasm_exec.js`, does any reason didn't implement it?

It seems can work correctly if I change the disabled code to below:

```js
// func valueInstanceOf(v ref, t ref) bool
"syscall/js.valueInstanceOf": (sp) => {
  return loadValue(sp) instanceof loadValue(sp + 8);
},
```

deadprogram, 2020-10-09:
Hello @elct9620 that sounds like it would be a nice pull request, unless someone else knows why it was commented out?

elct9620, 2020-10-09:
I think I can create a pull request first and wait for the comment if anyone knows about that.

deadprogram, 2020-10-09:
That would be very appreciated, thanks!

elct9620, 2020-10-09:
The pull request is here #1432, I will close this issue and let us move to PR to continue discuss.

closed 2020-10-09 by elct9620