tinygo-org/tinygo #1428: ESP8266 i2c support

sh0rez, 2020-10-07:
The docs (https://tinygo.org/microcontrollers/esp8266/) list that the ESP8266 i2c feature is not yet implemented.

To interface sensors like the BME280, I2C is required. What needs to be done so i2c can be used?

aykevl, 2020-10-27:
The ESP8266 unfortunately does not have hardware support for I2C. See https://github.com/tinygo-org/tinygo-site/pull/125.

aykevl, 2020-10-27:
To add support for I2C, a generic software I2C library needs to be written. I think the right place for this would be the machine package. Perhaps tinygo.org/x/drivers/bitbang with type `I2C`?

So you could use it like this:

```go

import "tinygo.org/x/drivers/bitbang"
import "tinygo.org/x/drivers/bme280"

func main() {
    i2c := bitbang.NewI2C(scl, sda)
    sensor := bme280.New(i2c)
    sensor.Configure(...)
    // ...
}
```

DanValenza, 2021-01-17:
I am interested in working on the I2C bitbang library, I've found [these specs](https://www.ti.com/lit/an/slva704/slva704.pdf) for the I2C protocol, are there stable implementations in other languages out there we could refer to?

DanValenza, 2021-01-17:
I did a bit of digging and [this](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_si2c.cpp) is from the Arduino core library for ESP8266

cweagans, 2021-02-03:
https://github.com/periph/devices/tree/main/bitbang may be of interest here. It might be more expedient to separate that implementation from the rest of the periph.io suite, but then again, adopting periph.io to enhance tinygo device support across the board might be a net gain?

deadprogram, 2021-11-18:
This was implemented in https://github.com/tinygo-org/drivers/pull/319 and was released. Now closing, thank you!

closed 2021-11-18 by deadprogram