tinygo-org/tinygo #1433: Slack Invitation

ewwaller, 2020-10-09:
The link for a Slack invitation in the Readme is dead.   
I would like to obtain an invite -- this is a cool project.

deadprogram, 2020-10-10:
The links I just tried worked. You do have to request an invitation however from https://invite.slack.golangbridge.org/ before you can access it.

ewwaller, 2020-10-11:
That link leads to a page that says  [quote]Slack
This link is no longer active
To join this workspace, you’ll need to ask the person who originally invited you for a new link.[/quote].

deadprogram, 2020-10-12:
There is a problem according to the Slack admins. More on this soon, hopefully.

ewwaller, 2020-10-30:
This seems to have been resolved.

deadprogram, 2020-10-30:
OK good. Now closing, thanks for updating us.

closed 2020-10-30 by deadprogram