tinygo-org/tinygo #1449: big.NewInt build failed

nobonobo, 2020-10-19:
bigint-issue.go
```go
package main

import "math/big"

var bigZero = big.NewInt(0)

func main() {}
```

 ```
$ tinygo build -o main ./bigint-issue.go 
# command-line-arguments
/usr/local/Cellar/go/1.15.2/libexec/src/math/big/nat.go:74:11: interp: unknown GEP

traceback:
/usr/local/Cellar/go/1.15.2/libexec/src/math/big/nat.go:74:11:
  %16 = getelementptr inbounds i64, i64* %4, i64 0, !dbg !1658
/usr/local/Cellar/go/1.15.2/libexec/src/math/big/nat.go:84:19:
  %7 = call { i64*, i64, i64 } @"(math/big.nat).setWord"(i64* %4, i64 %5, i64 %6, i64 %x, i8* undef, i8* undef), !dbg !1663
/usr/local/Cellar/go/1.15.2/libexec/src/math/big/int.go:55:25:
  %17 = call { i64*, i64, i64 } @"(math/big.nat).setUint64"(i64* %14, i64 %15, i64 %16, i64 %2, i8* undef, i8* undef), !dbg !1671
/usr/local/Cellar/go/1.15.2/libexec/src/math/big/int.go:69:26:
  %2 = call %"math/big.Int"* @"(*math/big.Int).SetInt64"(%"math/big.Int"* %0, i64 %x, i8* undef, i8* undef), !dbg !1655
command-line-arguments/<init>:9:25:
  %0 = call %"math/big.Int"* @"math/big.NewInt"(i64 0, i8* undef, i8* undef), !dbg !1650
```

## Other Info.
```
$ go version
go version go1.15.2 darwin/amd64
```

```
$ tinygo version
tinygo version 0.15.0 darwin/amd64 (using go version go1.15.2 and LLVM version 10.0.1)
```

sago35, 2020-10-26:
I built it with the latest dev version.
It didn't build as well, with a slightly different error.

```
sago35@sago35-B450GT3  ~/tinygo/tinygo (dev)
$ tinygo-0.16.0-dev-ffeff55 version
tinygo version 0.16.0-dev-ffeff55 linux/amd64 (using go version go1.14.4 and LLVM version 10.0.1)

sago35@sago35-B450GT3  ~/tinygo/tinygo (dev)
$ tinygo-0.16.0-dev-ffeff55 build -o main ./src/examples/i1449-big-newint/
# math/big
/usr/local/go/src/math/bits.go: interp: branch on a non-constant

traceback:
/usr/local/go/src/math/bits.go:
  br i1 %1, label %if.then, label %if.done, !dbg !1768
/usr/local/go/src/math/frexp.go:26:20:
  %7 = call { double, i64 } @math.normalize(double %f, i8* undef, i8* undef), !dbg !1779
/home/sago35/tinygo/tinygo/src/runtime/math.go:129:62:
  %0 = call { double, i64 } @math.frexp(double %x, i8* undef, i8* undef), !dbg !1766
/usr/local/go/src/math/big/float.go:560:26:
  %38 = call { double, i64 } @math.Frexp(double %x, i8* undef, i8* undef), !dbg !1810
/usr/local/go/src/math/big/float.go:92:30:
  %3 = call %"math/big.Float"* @"(*math/big.Float).SetFloat64"(%"math/big.Float"* %2, double %x, i8* undef, i8* undef), !dbg !1770
math/big/<init>:10:18:
  %39 = call %"math/big.Float"* @"math/big.NewFloat"(double 3.000000e+00, i8* undef, i8* undef), !dbg !1830
```

sago35, 2020-10-28:
If the changes in #1430 are merged, the build will succeed.

```
sago35@sago35-B450GT3  ~/tinygo/tinygo (programmer-7)
$ cat ./src/examples/i1449-big-newint/main.go 
package main

import (
        "fmt"
        "math/big"
)

var bigZero = big.NewInt(0)

func main() {
        fmt.Printf("hello\n")
}

sago35@sago35-B450GT3  ~/tinygo/tinygo (programmer-7)
$ ./bin/tinygo-0.16.0-dev-539a81b version
tinygo version 0.16.0-dev-539a81b linux/amd64 (using go version go1.14.10 and LLVM version 10.0.1)

sago35@sago35-B450GT3  ~/tinygo/tinygo (programmer-7)
$ ./bin/tinygo-0.16.0-dev-539a81b build -o main ./src/examples/i1449-big-newint/main.go 

sago35@sago35-B450GT3  ~/tinygo/tinygo (programmer-7)
$ ./main 
hello
```

niaow, 2022-11-04:
Confirmed fixed.

closed 2022-11-04 by niaow