tinygo-org/tinygo #1450: undeclared name

ttttmr, 2020-10-19:
This is a simplified version of my code, there are two files

```go
//main.go
package main
...
func main() {
  somefunc()
}
```

```go
//utils.go
package main
...
func somefunc(){
  //do something
}
```

I use docker to compile

`docker run --rm -v `pwd`:/src -e "GOPATH=/go" tinygo/tinygo:0.15.0 /bin/bash -c "cd /src && tinygo build -o main.wasm -target wasm main.go"`

error:
```
# command-line-arguments
main.go:x:x: undeclared name: somefunc
```

main.go cannot find the function in utils.go

closed 2020-10-20 by ttttmr

jrauschenbusch, 2022-01-15:
@ttttmr How did you solved the issue? I'm facing the same one.

maddin200, 2022-03-22:
tinygo build ./ 

is definitely different from:

tinygo build ./utils.go ./main.go // not possible
 especially in comparison to the standard go compiler
go build ./utils.go ./main.go

flyn-org, 2023-11-17:
Would it be possible to support multiple input files? It would be nice to be able to use tinygo as a drop-in replacement to Google's Go compiler in this regard.