tinygo-org/tinygo #1472: cannot load runtime/debug: malformed module path "runtime/debug": missing dot in first path element

happybeing, 2020-10-31:
I'm trying to compile existing Go code ([git-bug](https://github.com/MichaelMure/git-bug/) and its dependencies) as part of a project to run them in the browser by compiling to Web Assembly. But as soon as TinyGo has downloaded the dependencies I get at error related to `runtime/debug`:

```
$ tinygo build -o git-bug
build github.com/MichaelMure/git-bug: cannot load runtime/debug: malformed module path "runtime/debug": missing dot in first path element
```
I'm new to Go/TinyGo and had difficulty finding `runtime/debug` (maybe its no longer current?) but with help found it mentioned in TinyGo's *deprecated* [Tinygo Import Report](https://github.com/trashhalo/tinygo-import-report/blob/master/Readme.md#runtimedebug), but it is not mentioned at all on the successor [Packages supported by TinyGo](https://tinygo.org/lang-support/stdlib/) mentioned in the 'Report'.

So I'm trying to clarify. `git-bug` is using Go 1.13 which is what is in the package repositories atm, so not *that* old. Any explanations or hints on the cause and how to resolve are welcome. 

Searching I found very little about porting Go to TinyGo, so if there are any guides to this please share. Also knowing which version of Go is most closely matched to TinyGo would be useful.

Thanks. In the meantime I'll stay with using Go/Wasm for now and come back to TinyGo later.

aykevl, 2020-10-31:
I tried this myself and I got a much better error:

```
../../pkg/mod/github.com/stretchr/testify@v1.6.1/assert/assertions.go:14:2: package runtime/debug is not in GOROOT (/home/ayke/.cache/tinygo/goroot-go1.14-5e22ce3d92ccb315b09d6b235aa0d844cca67e0d4b1cdd900a7fb3bca52002db/src/runtime/debug)
package github.com/MichaelMure/git-bug
	imports github.com/MichaelMure/git-bug/commands
	imports github.com/MichaelMure/git-bug/api/graphql
	imports github.com/99designs/gqlgen/graphql/handler
	imports github.com/99designs/gqlgen/graphql
	imports runtime/debug
```

The runtime/debug package is part of the GOROOT and is part of the Go runtime (although not part of the Go runtime package). In this case you can see the dependency list: runtime/debug is imported by the testify package.

To be honest, I think it's unlikely this program will work in TinyGo soon considering the massive list of dependencies. The runtime/debug is one thing that's not implemented but it's unlikely to be the last.

aykevl, 2020-10-31:
> Searching I found very little about porting Go to TinyGo, so if there are any guides to this please share. Also knowing which version of Go is most closely matched to TinyGo would be useful.

TinyGo tries to be compatible with all Go version from Go 1.11 to 1.15. For WebAssembly specifically we had to make an incompatible change in wasm_exec.js which means for WebAssembly only Go 1.14-1.15 is supported, but that's the only exception.

deadprogram, 2020-12-27:
Since we are not able to support WebAssembly in older versions of Go, I am going to close this issue. Please reopen if needed.

closed 2020-12-27 by deadprogram

happybeing, 2020-12-27:
This isn't just an issue with older Go versions though I was originally thinking I should be using go 1.13, that was my mistake (it's the minimum version). Using go 1.15 the error is as reported by @aykevl:
```
$ tinygo build -o git-bug
go: downloading github.com/fatih/color v1.9.0
go: downloading github.com/mattn/go-colorable v0.1.4
../../../.gvm/pkgsets/go1.15/global/pkg/mod/github.com/stretchr/testify@v1.6.1/assert/assertions.go:14:2: package runtime/debug is not in GOROOT (/home/mrh/.cache/tinygo/goroot-go1.15-493b96244d98e1b984bfbf6f5daef9bb7bc06c82dbf6aaf0b9d47fd98556958b/src/runtime/debug)
package github.com/MichaelMure/git-bug
	imports github.com/MichaelMure/git-bug/commands
	imports github.com/MichaelMure/git-bug/api/graphql
	imports github.com/99designs/gqlgen/graphql/handler
	imports github.com/99designs/gqlgen/graphql
	imports runtime/debug
```

I'm not saying keep this issue open unless it is useful to you guys, just want to point out that it isn't solved by using the latest Go version.

FYI Given the pessimism here about it being able to compile git-bug with TinyGo we've decided to look for alternatives to Go for this in the long term. I successfully built a proof-of-concept app ([p2p Git Portal](https://github.com/happyborg/p2p-git-portal-poc)) using Go/Wasm, but given the 12MB runtime overhead Go is not really an option for web app so I'm looking at options to switch to Rust/Wasm. It will be more work, as we'll have to re-write anything not available such as git-bug, but I feel more confident about Rust/Wasm support and performance than I do Go/Wasm or TinyGo/Wasm at the moment. I've liked learning a bit of Go and getting to know the community, so it's been a good and useful experience.