tinygo-org/tinygo #1478: "nrf51" is wrong "openocd-target" for nrf52840 : should be nrf52

captcha1, 2020-11-02:
the "openocd-target" for nrf52840 should be nrf52 : 

$ diff  /usr/lib64/tinygo/targets/nrf52840.json*
16c16
<       "openocd-target": "nrf52"
---
>       "openocd-target": "nrf51"

captcha1, 2020-11-02:
Oops : 
```

$ diff  /usr/lib64/tinygo/targets/nrf52840.json*
16c16
<       "openocd-target": "nrf52"
---
>       "openocd-target": "nrf51"

```

aykevl, 2020-11-02:
This is intentional, as older OpenOCD versions don't understand the nrf52 target and the nrf51 target seems to work just as well.
Is this `openocd-target` causing a problem?

captcha1, 2020-11-03:
Well ... I tried it on a different laptop (Fedora 33, openocd-0.10.0) and it works OK w/ nrf51 and fails w/ nrf52 : 
```

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.0
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : reduce speed request: 10000kHz to 5000kHz maximum
Info : clock speed 10000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000002bc msp: 0x20010000
** Programming Started **
** Programming Failed **
shutdown command invoked
embedded:startup.tcl:523: Error: 
in procedure 'flash' called at file "embedded:startup.tcl", line 523

```

Strange.

aykevl, 2020-11-03:
In that case it looks like nrf51 is the way to go, even if it seems strange for a nrf52840.

Remember these chips are mostly backwards compatible and the nrf51 was introduced much earlier. It's quite likely that support for the nrf51 also works for the nrf52 series.

closed 2020-11-03 by aykevl