tinygo-org/tinygo #1483: SPI1.Configure crashes for pygamer

jypelle, 2020-11-04:
I want to use TFT on pygamer, but calling machine.SPI1.Configure crash

```go
	machine.SPI1.Configure(machine.SPIConfig{
		SCK:       machine.SPI1_SCK_PIN,
		SDO:       machine.SPI1_SDO_PIN,
		SDI:       machine.SPI1_SDI_PIN,
	})
```

jypelle, 2020-11-04:
Found the solution thanks to @ardnew : "it looks like there is a bug in (SPI).Configure for the SAMD51 (src/machine/machine_atsamd51.go). the pygamer sets SPI1_SDI_PIN to NoPin (0xFF). when findPinMapping gets called with this pin (line 1334), it  tries to evaluate bytes := pinPadMapping[pin/2], i.e., pinPadMapping[255/2], and pinPadMapping only has 64 elements. that would be an access violation"

Some updates to do in src/machine/machine_atsamd51.go:

Call findPinPadMapping only when config.SDI != NoPin

```go
// Determine the input pinout (for SDI).
	var dataInPinout uint32
	var SDIPinMode PinMode
	if config.SDI != NoPin {
		var ok bool
		SDIPinMode, dataInPinout, ok = findPinPadMapping(spi.SERCOM, config.SDI)
		if !ok {
			return ErrInvalidInputPin
		}
	}
```

sago35, 2020-11-04:
#1408

sago35, 2020-11-04:
PR: #1485

sago35, 2020-11-04:
This will fix the crash problem with pybadge and pygamer.

commit db27541b (2020-11-08): Fix #1483
closed 2020-11-18 by aykevl