tinygo-org/tinygo #1486: Error compiling starlark-go to wasm

maxmcd, 2020-11-06:
Command to reproduce:
```
docker run -it tinygo/tinygo:0.15.0 bash -c "apt-get update && apt-get install -y git  && go get github.com/wasabi/starlark-go/cmd/demo && cd /root/go/src/github.com/wasabi/starlark-go && tinygo build -o wasm.wasm -target wasm ./cmd/demo/demo.go"
```

I'm using a fork at [github.com/wasabi/starlark-go](github.com/wasabi/starlark-go) that has minor fixes to remove features that tinygo doesn't support, but can't seem to figure out what this issue is.

Full Error:
```
tinygo: /home/circleci/project/llvm-project/llvm/include/llvm/Support/Casting.h:264: typename cast_retty<X, Y *>::ret_type llvm::cast(Y *) [X = llvm::ConstantExpr, Y = llvm::Value]: Assertion `isa<X>(Val) && "cast<Ty>() argument of incompatible type!"' failed.
SIGABRT: abort
PC=0x7f148517e7bb m=8 sigcode=18446744073709551610

goroutine 0 [idle]:
runtime: unknown pc 0x7f148517e7bb
stack: frame={sp:0x7f1456ffcb00, fp:0x0} stack=[0x7f14567fd248,0x7f1456ffce48)
00007f1456ffca00:  0000003100000049  00007f1438000bf8
00007f1456ffca10:  000000000000065f  0000ffff00001fa2
00007f1456ffca20:  ffffffffffffff70  0000000000000019
00007f1456ffca30:  0000000000000064  0000000000000001
00007f1456ffca40:  0000000000000007  ea12f36e66f0e800
00007f1456ffca50:  0000007700000007  00007f142c243ac0
00007f1456ffca60:  00007f142c243c00  00007f1438000020
00007f1456ffca70:  0000000000000004  0000000000000110
00007f1456ffca80:  0000000000000140  00007f14851cacbe
00007f1456ffca90:  0000000000000000  0000006e0000005b
00007f1456ffcaa0:  00007f142c243ad0  0000000000000106
00007f1456ffcab0:  00007f142c243ac0  0000000000000106
00007f1456ffcac0:  0000000000000110  00007f14851cbf3b
00007f1456ffcad0:  343632ffffffff70  00007f1438000020
00007f1456ffcae0:  00007f1456ffcb20  0000000000000105
00007f1456ffcaf0:  0000000000000105  00007f1456ffcd60
00007f1456ffcb00: <0000000000000000  00007f142c243ad0
00007f1456ffcb10:  0000000000000100  00007f14851bf477
00007f1456ffcb20:  00007f14fbad8000  00007f142c243ad0
00007f1456ffcb30:  00007f142c243b35  00007f142c243ad0
00007f1456ffcb40:  00007f142c243ad0  00007f142c243bd5
00007f1456ffcb50:  00007f142c243bfc  00007f142c243ad0
00007f1456ffcb60:  00007f142c243bfc  0000000000000000
00007f1456ffcb70:  0000000000000000  0000000000000000
00007f1456ffcb80:  fffffffe7fffffff  ffffffffffffffff
00007f1456ffcb90:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcba0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbb0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbc0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbd0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbe0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbf0:  ffffffffffffffff  ffffffffffffffff
runtime: unknown pc 0x7f148517e7bb
stack: frame={sp:0x7f1456ffcb00, fp:0x0} stack=[0x7f14567fd248,0x7f1456ffce48)
00007f1456ffca00:  0000003100000049  00007f1438000bf8
00007f1456ffca10:  000000000000065f  0000ffff00001fa2
00007f1456ffca20:  ffffffffffffff70  0000000000000019
00007f1456ffca30:  0000000000000064  0000000000000001
00007f1456ffca40:  0000000000000007  ea12f36e66f0e800
00007f1456ffca50:  0000007700000007  00007f142c243ac0
00007f1456ffca60:  00007f142c243c00  00007f1438000020
00007f1456ffca70:  0000000000000004  0000000000000110
00007f1456ffca80:  0000000000000140  00007f14851cacbe
00007f1456ffca90:  0000000000000000  0000006e0000005b
00007f1456ffcaa0:  00007f142c243ad0  0000000000000106
00007f1456ffcab0:  00007f142c243ac0  0000000000000106
00007f1456ffcac0:  0000000000000110  00007f14851cbf3b
00007f1456ffcad0:  343632ffffffff70  00007f1438000020
00007f1456ffcae0:  00007f1456ffcb20  0000000000000105
00007f1456ffcaf0:  0000000000000105  00007f1456ffcd60
00007f1456ffcb00: <0000000000000000  00007f142c243ad0
00007f1456ffcb10:  0000000000000100  00007f14851bf477
00007f1456ffcb20:  00007f14fbad8000  00007f142c243ad0
00007f1456ffcb30:  00007f142c243b35  00007f142c243ad0
00007f1456ffcb40:  00007f142c243ad0  00007f142c243bd5
00007f1456ffcb50:  00007f142c243bfc  00007f142c243ad0
00007f1456ffcb60:  00007f142c243bfc  0000000000000000
00007f1456ffcb70:  0000000000000000  0000000000000000
00007f1456ffcb80:  fffffffe7fffffff  ffffffffffffffff
00007f1456ffcb90:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcba0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbb0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbc0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbd0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbe0:  ffffffffffffffff  ffffffffffffffff
00007f1456ffcbf0:  ffffffffffffffff  ffffffffffffffff

goroutine 1 [syscall]:
runtime.cgocall(0xd39310, 0xc008d3e5a0, 0xd3b260)
	/usr/local/go/src/runtime/cgocall.go:133 +0x5b fp=0xc008d3e570 sp=0xc008d3e538 pc=0x9812eb
tinygo.org/x/go-llvm._Cfunc_LLVMGetConstOpcode(0x7f1438a0d3c8, 0x7f1400000000)
	_cgo_gotypes.go:5498 +0x49 fp=0xc008d3e5a0 sp=0xc008d3e570 pc=0xc13fe9
tinygo.org/x/go-llvm.Value.Opcode.func1(0x7f1438a0d3c8, 0xc008d3e608)
	/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/ir.go:899 +0x4d fp=0xc008d3e5d0 sp=0xc008d3e5a0 pc=0xc3626d
tinygo.org/x/go-llvm.Value.Opcode(0x7f1438a0d3c8, 0x0)
	/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/ir.go:899 +0x2b fp=0xc008d3e5f0 sp=0xc008d3e5d0 pc=0xc2316b
github.com/tinygo-org/tinygo/interp.(*MapValue).PutString(0xc008722d20, 0xc008d3ede8, 0xc008d3edd8, 0xc00c40e370, 0x1, 0x7f142c243708)
	/home/circleci/project/interp/values.go:335 +0x49 fp=0xc008d3e670 sp=0xc008d3e5f0 pc=0xcd75e9
github.com/tinygo-org/tinygo/interp.(*frame).evalBasicBlock(0xc008d3f130, 0x7f1439426db0, 0x7f1439426980, 0x0, 0x0, 0x0, 0x0, 0xc00c311ba0, 0x1, 0x1, ...)
	/home/circleci/project/interp/frame.go:318 +0x9cae fp=0xc008d3f0a0 sp=0xc008d3e670 pc=0xcd000e
github.com/tinygo-org/tinygo/interp.(*evalPackage).function(0xc008d3f340, 0x7f143005efd8, 0xc008d3f358, 0x2, 0x2, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/circleci/project/interp/interp.go:113 +0x238 fp=0xc008d3f258 sp=0xc008d3f0a0 pc=0xcd3888
github.com/tinygo-org/tinygo/interp.Run(0x908f720, 0x0, 0x0, 0x0)
	/home/circleci/project/interp/interp.go:85 +0x7fe fp=0xc008d3f4c8 sp=0xc008d3f258 pc=0xcd32ce
github.com/tinygo-org/tinygo/builder.Build(0x7fff02de0eab, 0x12, 0x7fff02de0e94, 0x9, 0xc0002ca4b0, 0xc008d3fba8, 0x0, 0x0)
	/home/circleci/project/builder/build.go:65 +0x210 fp=0xc008d3fb68 sp=0xc008d3f4c8 pc=0xd04940
main.Build(0x7fff02de0eab, 0x12, 0x7fff02de0e94, 0x9, 0xc0002ce000, 0x0, 0x12)
	/home/circleci/project/main.go:94 +0xc5 fp=0xc008d3fbd0 sp=0xc008d3fb68 pc=0xd17775
main.main()
	/home/circleci/project/main.go:856 +0x1f96 fp=0xc008d3ff88 sp=0xc008d3fbd0 pc=0xd1c136
runtime.main()
	/usr/local/go/src/runtime/proc.go:203 +0x1fa fp=0xc008d3ffe0 sp=0xc008d3ff88 pc=0x9b39aa
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:1373 +0x1 fp=0xc008d3ffe8 sp=0xc008d3ffe0 pc=0x9e09a1

rax    0x0
rbx    0x6
rcx    0x7f148517e7bb
rdx    0x0
rdi    0x2
rsi    0x7f1456ffcb00
rbp    0x7f14852cbee0
rsp    0x7f1456ffcb00
r8     0x0
r9     0x7f1456ffcb00
r10    0x8
r11    0x246
r12    0x53e963d
r13    0x53e9685
r14    0x108
r15    0x100
rip    0x7f148517e7bb
rflags 0x246
cs     0x33
fs     0x0
gs     0x0
```

aykevl, 2020-11-06:
That's indeed an important bug. However, the package this is coming from has largely been rewritten in #1430 and this PR should be merged soon. That PR should remove the code responsible for the bug and is thus likely to fix it.

maxmcd, 2020-11-06:
@aykevl nice! I tested it, it works! thanks, closing

closed 2020-11-06 by maxmcd