tinygo-org/tinygo #1493: Working with Tessel board

hajsf, 2020-11-11:
I've tessel board (tessel 1 not tessel 2) https://tessel.io/community

can I use tinygo with it?

aykevl, 2020-11-11:
I've taken a quick look at it. There appear to be two chips:

* A big ARM SoC that runs user code (and I guess runs Linux). If you want to use Go on it, I would suggest you use regular Go, not TinyGo.
* A smaller SAMD21 chip. This one is in fact supported by TinyGo, but I suspect it is not meant to be programmed directly.

How do you want to use TinyGo on the Tessel?

hajsf, 2020-11-12:
I've the tessel 1 board and tested it with node, just came across tinyGO and thought to give it a try on this board.

deadprogram, 2020-11-12:
The most interesting part was the work on the LuaJIT compiler, IMO.

In any case, it is pretty old board, and not all that many were manufactured. So probably not very easy to add TinyGo support.

closed 2020-11-14 by hajsf