tinygo-org/tinygo #1496: Arduino: image/color RGBA() function breaks build

Nerzal, 2020-11-14:
Hi i narrowed down, why display drivers like st7735 won't build on the Arduino UNO. 
It is the RGBA function call. 

`func (c RGBA) RGBA() (r, g, b, a uint32) {
	r = uint32(c.R)
	r |= r << 8
	g = uint32(c.G)
	g |= g << 8
	b = uint32(c.B)
	b |= b << 8
	a = uint32(c.A)
	a |= a << 8
	return
}`

This results in a SIGSEV during build

```
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x7f7278655260 pc=0x7f62ae44e11f]

runtime stack:
runtime.throw(0x7f823c, 0x2a)
        /usr/local/go/src/runtime/panic.go:1116 +0x72
runtime.sigpanic()
        /usr/local/go/src/runtime/signal_unix.go:704 +0x4ac

goroutine 1 [syscall]:
runtime.cgocall(0x74c530, 0xc00263f3e8, 0xc002239570)
        /usr/local/go/src/runtime/cgocall.go:133 +0x5b fp=0xc00263f3b8 sp=0xc00263f380 pc=0x420e1b
tinygo.org/x/go-llvm._Cfunc_LLVMTargetMachineEmitToMemoryBuffer(0xd42aa0, 0xd5fea0, 0x1, 0xc002239550, 0xc002239570, 0xc000000000)
        _cgo_gotypes.go:9296 +0x4d fp=0xc00263f3e8 sp=0xc00263f3b8 pc=0x667f6d
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer.func1(0xd42aa0, 0xd5fea0, 0xc000000001, 0xc002239550, 0xc002239570, 0x1b)
        /home/tobias/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0xff fp=0xc00263f428 sp=0xc00263f3e8 pc=0x6859bf
tinygo.org/x/go-llvm.TargetMachine.EmitToMemoryBuffer(0xd42aa0, 0xd5fea0, 0x1, 0xc002a848c0, 0x1b, 0x14)
        /home/tobias/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200503225853-345b2947b59d/target.go:273 +0x85 fp=0xc00263f490 sp=0xc00263f428 pc=0x6734e5
github.com/tinygo-org/tinygo/builder.Build(0x7ffe63c1ecc2, 0x17, 0x7e52d8, 0x4, 0xc000252390, 0xc00263fb78, 0x0, 0x0)
        /home/tobias/go/src/github.com/Nerzal/tinygo/builder/build.go:166 +0x6bf fp=0xc00263fb10 sp=0xc00263f490 pc=0x72b43f
main.Flash(0x7ffe63c1ecc2, 0x17, 0x0, 0x0, 0xc000220000, 0x0, 0x15)
        /home/tobias/go/src/github.com/Nerzal/tinygo/main.go:188 +0x1a5 fp=0xc00263fbc0 sp=0xc00263fb10 pc=0x73c5e5
main.main()
        /home/tobias/go/src/github.com/Nerzal/tinygo/main.go:894 +0xfb9 fp=0xc00263ff88 sp=0xc00263fbc0 pc=0x73fc59
runtime.main()
        /usr/local/go/src/runtime/proc.go:204 +0x209 fp=0xc00263ffe0 sp=0xc00263ff88 pc=0x454c69
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1374 +0x1 fp=0xc00263ffe8 sp=0xc00263ffe0 pc=0x4865a1
```

Is there anything we could do to work arround this issue?

deadprogram, 2024-12-05:
@Nerzal is this still an issue with the latest TinyGo versions or can this be closed?

davecheney, 2026-10-26:
The following sample compiles and runs on an Arduino Uno. It repeatedly prints the expected RGBA channel values over serial at 9600 baud.

```go
package main

import (
	"image/color"
	"time"
)

//go:noinline
func channels(c color.RGBA) (uint32, uint32, uint32, uint32) {
	return c.RGBA()
}

func main() {
	for {
		r, g, b, a := channels(color.RGBA{
			R: 0x12, G: 0x34, B: 0x56, A: 0xff,
		})
		println(r, g, b, a)
		time.Sleep(time.Second)
	}
}
```

Save as `main.go`, then flash and monitor:

```sh
tinygo flash -target=arduino -monitor -baudrate=9600 main.go
```

Observed output:

```text
4626 13364 22102 65535
```

This verifies `color.RGBA.RGBA()` on Uno hardware, not a display driver.

closed 2026-10-26 by davecheney