tinygo-org/tinygo #1503: esp32: No available targets are compatible with triple "xtensa"

jbpratt, 2020-11-19:
Hi! I am super excited about ESP32 and ESP8266 support landing recently.

I think something is misconfigured on my side, it seems I can't flash the examples. I am currently working on a learning project so I do know that using `idf.py` and the traditional(?) cmake set up from espressif, I am able to flash the controller without issue.

I am quite new to this, `esptool.py` may simply be a wrapper around `idf.py` or the other way around, so I apologize if I'm not making much sense.

The exact board I am working with is [this](https://www.amazon.com/KeeYees-Development-Bluetooth-Microcontroller-ESP-WROOM-32/dp/B07QCP2451)

```
autumns-esp32 >tinygo flash -target=esp32-wroom-32 -port=/dev/ttyUSB0 examples/blinky1
error: No available targets are compatible with triple "xtensa"

autumns-esp32 >esptool.py flash_id
esptool.py v2.9-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting........_____....._____.....____
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:aa:b5:8b:d4:30
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

autumns-esp32 >tinygo version
tinygo version 0.15.0 linux/amd64 (using go version go1.15.5 and LLVM version 11.0.0)

autumns-esp32 >esptool.py version
esptool.py v2.9-dev
2.9-dev
```

deadprogram, 2020-11-20:
Hello @jbpratt we have a small delay in updating the docs. The command you need for flashing is:

```
tinygo flash -target=esp32-coreboard-v2 -port=/dev/ttyUSB0 examples/blinky1
```

or 

```
tinygo flash -target=esp32-mini32 -port=/dev/ttyUSB0 examples/blinky1
```

depending on which board you have.

Hope that helps!

jbpratt, 2020-11-22:
Thanks for the quick reply. It seems I'm still having the issue on `0.16.0`

```
tmp >tinygo flash -target=esp32 -port=/dev/ttyUSB0 examples/blinky1
error: No available targets are compatible with triple "xtensa"
tmp >tinygo version
tinygo version 0.16.0 linux/amd64 (using go version go1.15.5 and LLVM version 11.0.0)
```

Do note the LLVM version, which I think is higher than what is expected. I think this may be the culprit https://github.com/tinygo-org/tinygo/issues/1435#issuecomment-710365072 but @aykevl or @anatol may be able to provide clarity. Sorry about the confusion. Feel free to close this and I will try to hack around without using the AUR distribution for now.

deadprogram, 2020-11-22:
Pretty sure you need a more specific target than just `esp32`. Try either `esp32-coreboard-v2` or else `esp32-mini32`.

aykevl, 2020-11-22:
This error:

    error: No available targets are compatible with triple "xtensa"

Means that the LLVM this TinyGo version was built with, doesn't support the Xtensa backend which is used for the ESP32 and ESP8266. To fix this, you either need to download one of the release binaries or build TinyGo and LLVM yourself (see BUILDING.md), distributions don't use the Xtensa fork of LLVM.

jbpratt, 2020-11-22:
@deadprogram I attempted with all of them, but no luck. @aykevl Thank you, I will adjust my setup.

```
~ >tinygo targets | grep esp32
esp32-coreboard-v2
esp32-mini32
esp32
```

Thank you both for the help!!

closed 2020-11-22 by jbpratt

anatol, 2020-11-29:
> Means that the LLVM this TinyGo version was built with, doesn't support the Xtensa backend which is used for the ESP32 and ESP8266. To fix this, you either need to download one of the release [...]

Arch Linux ships the upstream LLVM. Is `xtensa` support part of upstream LLVM now? If not then [a custom LLVM](https://aur.archlinux.org/packages/llvm-xtensa-git/) needs to be built by the end-user.

cc @foutrelis

aykevl, 2020-12-01:
Unfortunately, the Xtensa backend is not upstream in LLVM.

However, I believe at least one Arch Linux TinyGo package uses builds published by us (at https://github.com/tinygo-org/tinygo/releases) which do include Xtensa support.