tinygo-org/tinygo #1504: WebAssembly binary built targeting the WASI interface cannot read environment variables

flavio, 2020-11-20:
I looks like `os.Getenv` is not working for WASM modules produced for the WASI target.

# How to reproduce the issue

Given the following program called `env-test.go`:

```go
package main

import (
	"fmt"
	"os"
)

func main() {
	fmt.Printf("HOME -> |%s|\n", os.Getenv("HOME"))
}
```

Build the program using latest (0.16.0) release of TinyGo:

```bash
$ tinygo build -o env-test.wasm -wasm-abi=generic -target=wasi env-test.go
```

And then run it with `wasmtime`:

```bash
$ wasmtime run --env HOME=flavio env-test.wasm
HOME -> ||
```

As you can see the `os.Getenv` returns an empty string.

# Environment details

* tinygo version 0.16.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1)
* wasmtime 0.21.0

flavio, 2020-11-20:
I would like to help fixing this issue, is there anybody that can give me some guidance? I tried to understand how the WASI target is implemented internally, but I didn't succeed (yet) :heart:

aykevl, 2020-12-02:
Yes, certainly I would like to help. Do you have specific things you would like to know? (See https://github.com/tinygo-org/tinygo/issues/1505#issuecomment-736872118).

mathetake, 2021-02-16:
Here's the PR to solve this: https://github.com/tinygo-org/tinygo/pull/1632

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram