tinygo-org/tinygo #1505: WebAssembly binary built targeting the WASI interface cannot read from standard input

flavio, 2020-11-20:
Reading from standard input from a Go binary compiled for the WASI interface is not working.
It seems the WASI backend of TinyGo doesn't have `fd_read` implemented.

# How to reproduce the issue

Create a simple program named `stdin-test.go`:

```go
package main

import (
	"bufio"
	"fmt"
	"os"
)

func main() {
	reader := bufio.NewReader(os.Stdin)
	fmt.Println("say something")
	msg, err := reader.ReadString('\n')
	if err != nil {
		fmt.Printf("I got an error: %v\n", err)
	} else {
		fmt.Printf("This is what I got: |%s|\n", msg)
	}
}
```

Build the program:

```bash
$ tinygo build -o stdin-test.wasm -wasm-abi=generic -target=wasi stdin-test.go
```

Execute the program:

```bash
$ wasmtime run stdin-test.wasm 
say something
I got an error: read /dev/stdin: operation not supported
```

# Environment details

* tinygo version 0.16.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1)
* wasmtime 0.21.0

flavio, 2020-11-20:
By looking at the code I see `fd_read` support is not yet done (there's even a `TODO`). I tried to add it, but failed... I would like to help fixing this issue, but I would need some generic guidance :heart:

pojntfx, 2020-11-24:
I'm currently porting my [Berkeley sockets for WebAssembly via WebRTC](https://github.com/pojntfx/webassembly-berkeley-sockets-via-webrtc) to TinyGo and I'd greatly appreciate this as well as I'm currently using stdin/stdout in the demo programs which does not seem to be supported.

Environment:

- tinygo version 0.15.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1)
- Node.js v12.19.0

trevor403, 2020-11-25:
You can get stdin using wasi if you implement the correct import bindings. I can work on getting support in xterm.js and make a PR.

mathetake, 2020-11-25:
Hi folks, I'm the author of original WASI target PR. so please ping me when you have the PR ready for review! would love to see this PR

aykevl, 2020-12-01:
Yes, this is something that isn't yet implemented. It will likely require some plumbing in the os package.

You should be able to call WASI functions using `//go:wasm-module` and `//export`, as long as the signatures match. This is already done in the runtime. If you need some help, feel free to ask.

pojntfx, 2020-12-06:
While not strictly tied to WASI (`target=wasm`), here is how I've polyfilled this in the browser for now in unisockets which might be of use: https://github.com/pojntfx/unisockets/blob/103149d62ae029dfc5dfea9d8414f242ca4cb208/cmd/unisockets_runner_web/main.ts#L310-L326

mathetake, 2021-02-16:
Update: I've started working on this issue, and the PR is almost there: https://github.com/mathetake/tinygo/compare/wasi-env-vars...mathetake:libc-wasi?expand=1

mathetake, 2021-02-24:
Update: I just created the PR to support basic file i/o: https://github.com/tinygo-org/tinygo/pull/1659

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

PR #1659 (2021-03-28): WASI & darwin: support basic file io based on libc
This is the first PR towards file system support in libc target.

resolves #1505
closed 2021-05-12 by deadprogram