tinygo-org/tinygo #1507: esp32: gdb support

jbpratt, 2020-11-22:
I was digging in a bit to determine how `gdb` support worked for existing controllers. Adding the field to the target file gets me a step in the right direction (though I am sure this has already been worked through and determined to delay).

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#command-line

```
+  "gdb": "xtensa-esp32-elf-gdb"
```

First error
```
tinygo >tinygo gdb -target=esp32
# os/user
../../../usr/lib/go/src/os/user/cgo_lookup_unix.go:19:6: not implemented: build constraints in #cgo line
```

which is a Solaris build tag. For the heck of it, I went ahead and deleted the build tag to see what would happen next and we seem to be getting into the meat of `go-llvm` and `llvm`
```
tinygo >tinygo gdb -target=esp32
# tinygo.org/x/go-llvm
../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/llvm_config_linux.go:8:9: invalid #cgo line: CPPFLAGS
../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/llvm_config_linux.go:9:9: invalid #cgo line: CXXFLAGS
../../../usr/include/llvm/Config/llvm-config.h:4:10: fatal: 'llvm-config-32.h' file not found
../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/executionengine.go:17:10: note: in file included from /home/jbpratt/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/executionengine.go!cgo.c:3:
../../../usr/include/llvm-c/ExecutionEngine.h:23:10: note: in file included from /usr/include/llvm-c/ExecutionEngine.h:23:
../../../usr/include/llvm-c/Target.h:24:10: note: in file included from /usr/include/llvm-c/Target.h:24:
../../../usr/include/llvm/Config/llvm-config.h:4:10: fatal: 'llvm-config-32.h' file not found
../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/target.go:17:10: note: in file included from /home/jbpratt/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/target.go!cgo.c:3:
../../../usr/include/llvm-c/Target.h:24:10: note: in file included from /usr/include/llvm-c/Target.h:24:
../../../usr/include/llvm/Config/llvm-config.h:4:10: fatal: 'llvm-config-32.h' file not found
../go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/version.go:16:10: note: in file included from /home/jbpratt/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20201104183921-570e7a6841d9/version.go!cgo.c:2:
```

```
tinygo >tinygo version
tinygo version 0.16.0 linux/amd64 (using go version go1.15.5 and LLVM version 10.0.1)
```

Please let me know if I can try to provide any more info or even help out with this. Thank you!

aykevl, 2020-11-23:
>     $ tinygo gdb -target=esp32
>     # os/user
>     ../../../usr/lib/go/src/os/user/cgo_lookup_unix.go:19:6: not implemented: build constraints in #cgo line

The problem here is that you are not specifying a package, so are trying to build the current package (which might be the tinygo compiler itself). The command `tinygo gdb` doesn't just connect using GDB, it compiles and flashes a program just like `tinygo flash` and tries to debug that. In this case you're getting a compiler error, so that error can't be related to debugging and would also be present with `tinygo flash`.

What you are probably looking for is this:

    tinygo gdb -target=esp32-mini32 examples/blinky1

But still, that won't work right away. I never attempted to get this working. What I did was compile an ELF file, flash it manually, and then run all commands manually to debug (`openocd` and `xtensa-esp32-elf-gdb`). Only _after_ you've done that, can you look into automating it with `tinygo gdb`.

However, I never found the JTAG access to the ESP32 to be very reliable. So what I did instead was that I ran the binary in [the Espressif fork of QEMU](https://github.com/espressif/qemu/wiki).

After you've compiled the QEMU fork, you can run a binary like this:

```sh
# create the image (only needs to be done once)
dd if=/dev/zero of=esp32.raw bs=1024 count=4096

# build and run the binary
tinygo build -o test.elf -target=esp32-mini32 examples/serial
esptool.py --chip=esp32 elf2image test.elf
dd if=test.bin of=esp32.raw bs=1 seek=4096 conv=notrunc
qemu-system-xtensa -machine esp32 -nographic -drive file=esp32.raw,if=mtd,format=raw
```

(note that `qemu-system-xtensa` points to the QEMU fork from Espressif)

This produces familiar output, all in QEMU (because it is running the same ROM as on the real chip):

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3ffaf000,len:116
load:0x40080000,len:676
entry 0x40080098
hello world!
hello world!
hello world!
hello world!
hello world!
hello world!
hello world!
```

You can then debug by instead adding `-s -S` at the end of the QEMU command and connecting to it using `xtensa-esp32-elf-gdb test.elf -ex 'target remote :1234'`.

If you really do want to use a JTAG connection, you can try this (with `openocd` being the Espressif version, of course):

```
openocd -f board/esp32-wrover-kit-3.3v.cfg
```

... and then connect to it using `xtensa-esp32-elf-gdb` in a separate window.

jbpratt, 2020-11-23:
> The problem here is that you are not specifying a package, so are trying to build the current package (which might be the tinygo compiler itself). The command tinygo gdb doesn't just connect using [...]

You are exactly correct, my dumb mistake. I will dig through this and see what I can do. Thank you for the detailed reply!

aykevl, 2022-04-22:
I have implemented GDB support for an ESP32 running inside QEMU: https://github.com/tinygo-org/tinygo/pull/2794

deadprogram, 2024-11-30:
Closing since some work on this was completed in #2794 thanks!

closed 2024-11-30 by deadprogram