tinygo-org/tinygo #1509: Are there any alternatives to net/http for get requests?

Nv7-GitHub, 2020-11-23:
I need to be able to send get requests using tinygo. Currently, the `net/http` package is not supported. Are there any alternatives to this package to sending GET requests?

aykevl, 2020-12-01:
None are supported right now but I have just come across this on Twitter that might be useful: https://twitter.com/pojntfx/status/1333563900575539200 (I did not check or test this).

sago35, 2021-09-28:
In baremetal, you can use tinygo.org/x/drivers/net.
An example is given below.

https://github.com/tinygo-org/drivers/tree/release/wifinina
https://github.com/tinygo-org/drivers/blob/3fe3fc64e4dedb899cd03542ef20d828a264045b/examples/wifinina/http-get/main.go#L77-L105

dosgo, 2021-09-28:
@sago35 Thanks! tinygo.org/x/drivers/net/http: module tinygo.org/x/drivers@latest found (v0.17.1), but does not contain package tinygo.org/x/drivers/net/http  This is because v0.17.1 does not?

sago35, 2021-09-28:
@dosgo 
Try the latest on the dev branch.

https://github.com/tinygo-org/drivers/tree/dev

deadprogram, 2024-11-27:
Closing since question seems to have been answered. Thank you!