tinygo-org/tinygo #1510: "undefined reference to time.startTimer" at a switch case <-time.After, for Arduino

chadmiller, 2020-11-24:
Use docker tinygo 0.15.0 to compile simple program

```go
package main

import (
        "time"
)

func main() {
        select {
                case <-time.After(time.Millisecond): {}
        }
}
```

```
docker run -v   `pwd`:/go -e "GOPATH=/go" tinygo/tinygo:latest env --chdir=/go tinygo build \
    -o /go/bugdemo -target arduino -scheduler tasks --no-debug pathto/bug/main.go
```

```
/tmp/tinygo855212794/main.o: In function `command-line-arguments.main':
main.go:(.text.command-line-arguments.main+0x96): undefined reference to `time.startTimer'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo855212794/main: exit status 1
```

I understand it's not implemented yet. 

The other bug I'm reporting is that this kind of thing should be in the FAQ.

aykevl, 2020-12-01:
> The other bug I'm reporting is that this kind of thing should be in the FAQ.

There are many things not yet implemented. It would be a massive amount to list them all, although this one is maybe frequent enough to warrant a FAQ entry.

However, see https://github.com/tinygo-org/tinygo/pull/1402 for some WIP code.

niaow, 2022-11-04:
Fixed by #2954

closed 2022-11-04 by niaow