tinygo-org/tinygo #1512: Failed to flash adafruit clue

rsjethani, 2020-11-27:
```
$ go version
go version go1.15.2 linux/amd64

$ tinygo version
tinygo version 0.15.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1)

$ ls
go.mod  go.sum  main.go

$ tinygo flash -target=clue_alpha main.go 
error: failed to flash /tmp/tinygo222558420/main.uf2: unable to locate device: FTHR840BOOT
```
Let me know if more context is required. Thank you!

deadprogram, 2020-11-28:
The target is now `clue-alpha` as of the 0.16.0 release, and we have standardized on the "-" across all targets. 

This line here in the target file for the Clue appears incorrect: https://github.com/tinygo-org/tinygo/blob/release/targets/clue-alpha.json#L6

It should probably be `CLUEBOOT`. If you change it in that file in your installation you should be able to verify that.

rsjethani, 2020-11-28:
@deadprogram thanks for your suggestion but even after upgrading tinygo and changing `FTHR840BOOT` to `CLUEBOOT` I am getting same error:
```
ryan@tuxedo:~/work/micro$ tinygo version
tinygo version 0.16.0 linux/amd64 (using go version go1.15.2 and LLVM version 10.0.1)
$ tinygo flash -target=clue-alpha main.go 
error: failed to flash /tmp/tinygo864220558/main.uf2: unable to locate device: CLUEBOOT
ryan@tuxedo:~/work/micro$
```
Just for the sake of completeness I was getting permission denied error earlier becuase my user was lacking `dialout` group permissions. I got a hint about this from:
```
$ ls -l /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 Nov 28 12:40 /dev/ttyACM0
```
After making my user a member of the dialout group things progressed but now I am stuck at the 'unable to locate device' error :(

deadprogram, 2020-11-28:
if you can get it into the bootloader mode by double clicking the reset button on the board, then the MSD drive should be mounted, and perhaps you can see what the name is?

rsjethani, 2020-11-28:
> if you can get it into the bootloader mode by double clicking the reset button on the board, then the MSD drive should be mounted, and perhaps you can see what the name is?
Here's the result:
```
$ mount | grep sdb
/dev/sdb on /media/ryan/CLUEBOOT type vfat (rw,nosuid,nodev,relatime,uid=1001,gid=1001,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
```
It correctly shows the name `CLUEBOOT` as you said earlier, I wonder what is the problem then?

deadprogram, 2020-11-28:
you can try this:

```
tinygo build -o /media/ryan/CLUEBOOT/flash.uf2 -target clue-alpha main.go
```

This command builds the UF2 binary right on to the boards MSD drive, which should cause the board to restart and run the tinygo program.

Subsequent flash command should kick the board into the bootoader, but will not be able to find the driver unless you have updated the target .json file.

If the .json file fix works, then perhaps you can submit a pull request to the `dev` branch with that change, if it is not too much trouble.

rsjethani, 2020-11-28:
@deadprogram I had to double click the reset button to as to get the MSD drive mounted. Then I simply used the flash command:
```
$ tinygo flash -target=clue-alpha test.go
```
It worked! From now on I do not have to mount the drive...simply connect the device and flash the program and I can immediately see the program running on device. This hint is right there on the [clue-alpha doc](https://tinygo.org/microcontrollers/clue-alpha/#troubleshooting).

But this wouldn't have worked without you pointing out the wrong volume name so thank you! I have created the PR accordingly.

rsjethani, 2020-11-28:
There is new behaviour which I think is not right:
When I flash the code, it does run immediately and shows the correct behaviour. However when I disconnect the device and try running it again by simply plugging it into my laptop's usb port or a power bank, the program I flashed doesn't run. I just see the builtin red led continuously on HIGH that is it. I tried the most basic program as given on the README.

aykevl, 2020-12-01:
Unfortunately I don't have this device so can't really test it.

I do know that another device (an ItsyBitsy M4 I have) sometimes seems to erase itself under circumstances I haven't figured out entirely, but that seem to be related to power. The bootloader remains in place though.
Can you reproduce this issue always or only sometimes?

rsjethani, 2020-12-02:
Always.

By looking at the dmesg logs I can say its not the ALPHA version...stil not 100%  sure. Also I see following dmesg logs when I simply connect the device to a USB port on my laptop:
```
[212681.152449] usb 1-1: device descriptor read/64, error -110
[212696.768376] usb 1-1: device descriptor read/64, error -110
[212697.004374] usb 1-1: new full-speed USB device number 10 using xhci_hcd

[212712.384437] usb 1-1: device descriptor read/64, error -110
[212728.000351] usb 1-1: device descriptor read/64, error -110
[212728.108345] usb usb1-port1: attempt power cycle
[212728.760231] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[212734.016377] usb 1-1: device descriptor read/8, error -110
[212739.392397] usb 1-1: device descriptor read/8, error -110
[212739.628212] usb 1-1: new full-speed USB device number 12 using xhci_hcd
[212744.768367] usb 1-1: device descriptor read/8, error -110
[212750.144385] usb 1-1: device descriptor read/8, error -110
[212750.252305] usb usb1-port1: unable to enumerate USB device
```
But after I double click the reset button to enter 'flashing' state, I can flash the program and immediatley after successful flash the device resets and starts running. Some dmesg logs during this process:

After double clicking:
```
[213611.640264] usb 1-1: new full-speed USB device number 19 using xhci_hcd
[213611.790738] usb 1-1: New USB device found, idVendor=239a, idProduct=0071, bcdDevice= 1.00
[213611.790744] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[213611.790748] usb 1-1: Product: Clue nRF52840
[213611.790751] usb 1-1: Manufacturer: Adafruit Industries
[213611.790754] usb 1-1: SerialNumber: XXXXX
[213611.793873] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[213611.795067] usb-storage 1-1:1.2: USB Mass Storage device detected
[213611.795472] scsi host2: usb-storage 1-1:1.2
[213612.805421] scsi host2: scsi scan: INQUIRY result too short (5), using 36
[213612.805430] scsi 2:0:0:0: Direct-Access     Adafruit nRF UF2          1.0  PQ: 0 ANSI: 2
[213612.806270] sd 2:0:0:0: Attached scsi generic sg1 type 0
[213612.806818] sd 2:0:0:0: [sdb] 8000 512-byte logical blocks: (4.10 MB/3.91 MiB)
[213612.807082] sd 2:0:0:0: [sdb] Write Protect is off
[213612.807087] sd 2:0:0:0: [sdb] Mode Sense: 03 00 00 00
[213612.807344] sd 2:0:0:0: [sdb] No Caching mode page found
[213612.807352] sd 2:0:0:0: [sdb] Assuming drive cache: write through
[213612.843806]  sdb:
[213612.847285] sd 2:0:0:0: [sdb] Attached SCSI removable disk
```
During and after flashing:
```
[213644.588150] usb 1-1: USB disconnect, device number 19
[213644.600640] blk_update_request: I/O error, dev sdb, sector 1 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[213644.600644] Buffer I/O error on dev sdb, logical block 1, lost async page write
[213644.600661] blk_update_request: I/O error, dev sdb, sector 33 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[213644.600662] Buffer I/O error on dev sdb, logical block 33, lost async page write
[213644.600668] blk_update_request: I/O error, dev sdb, sector 65 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[213644.600671] Buffer I/O error on dev sdb, logical block 65, lost async page write
[213644.705002] FAT-fs (sdb): unable to read boot sector to mark fs as dirty
[213644.908734] usb 1-1: new full-speed USB device number 20 using xhci_hcd
[213645.062069] usb 1-1: New USB device found, idVendor=239a, idProduct=8072, bcdDevice= 1.00
[213645.062083] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[213645.062087] usb 1-1: Product: Adafruit CLUE
[213645.062091] usb 1-1: Manufacturer: Adafruit
[213645.063668] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```

PS: I tried to use the board with circuit python....it works! I mean after flashing a python program...disconnecting...connecting to either a laptop usb port or a powerbank or even my phone charger works...the test python program runs. Not sure why it is not working with tinygo :(

deadprogram, 2020-12-03:
What is the vendor ID and product ID that appear in `dmesg` after you flash the Clue using Circuit Python? There is probably a mismatch with those values and what TinyGo is using for that board. I do not have a Clue (pun alert!) so I cannot test myself.

rsjethani, 2020-12-03:
I got following logs while using with circuit python:
```
[304134.125010] usb 1-1: new full-speed USB device number 28 using xhci_hcd
[304134.304887] usb 1-1: New USB device found, idVendor=239a, idProduct=8072, bcdDevice= 1.00
[304134.304893] usb 1-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[304134.304897] usb 1-1: Product: CLUE nRF52840 Express
[304134.304900] usb 1-1: Manufacturer: Adafruit Industries LLC
[304134.304902] usb 1-1: SerialNumber: XXXXX
[304134.312105] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[304134.320648] usb-storage 1-1:1.2: USB Mass Storage device detected
[304134.321005] scsi host2: usb-storage 1-1:1.2
[304134.365386] input: Adafruit Industries LLC CLUE nRF52840 Express Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8072.0002/input/input95
[304134.425971] input: Adafruit Industries LLC CLUE nRF52840 Express Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8072.0002/input/input96
[304134.426695] input: Adafruit Industries LLC CLUE nRF52840 Express Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8072.0002/input/input97
[304134.427082] input: Adafruit Industries LLC CLUE nRF52840 Express as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8072.0002/input/input98
[304134.427818] hid-generic 0003:239A:8072.0002: input,hidraw1: USB HID v1.11 Keyboard [Adafruit Industries LLC CLUE nRF52840 Express] on usb-0000:00:14.0-1/input3
[304134.428007] usbcore: registered new interface driver usbhid
[304134.428009] usbhid: USB HID core driver
[304134.433666] usbcore: registered new interface driver snd-usb-audio
[304135.330180] scsi host2: scsi scan: INQUIRY result too short (5), using 36
[304135.330191] scsi 2:0:0:0: Direct-Access     Adafruit CLUE nRF52840 Ex 1.0  PQ: 0 ANSI: 2
[304135.330904] sd 2:0:0:0: Attached scsi generic sg1 type 0
[304135.331457] sd 2:0:0:0: [sdb] 4089 512-byte logical blocks: (2.09 MB/2.00 MiB)
[304135.331740] sd 2:0:0:0: [sdb] Write Protect is off
[304135.331747] sd 2:0:0:0: [sdb] Mode Sense: 03 00 00 00
[304135.332013] sd 2:0:0:0: [sdb] No Caching mode page found
[304135.332025] sd 2:0:0:0: [sdb] Assuming drive cache: write through
[304135.366147]  sdb: sdb1
[304135.368306] sd 2:0:0:0: [sdb] Attached SCSI removable disk
```
From what I see vendor and product IDs are the same.

deadprogram, 2020-12-23:
I think something else is wrong with the nrf52840 reset implementation, as I am able to reproduce this problem on my itsybitsy-nrf52840 board. Looking into it.

rsjethani, 2020-12-23:
Ohh! ... and I was feeling stupid being a newbie microcontroller enthusiast and missing something very basic. Best of luck you, hope you could get some solution. The device does work with Python but TBH I don't want to code in Python

deadprogram, 2020-12-28:
Using https://github.com/tinygo-org/tinygo/pull/1548 and renaming the volume expected to `CLUEBOOT` and this is now working for me on the Clue I have recently obtained.

deadprogram, 2020-12-29:
#1549 has the correct volume name and just needs to be merged into `dev`.

#1548 has already been merged.

Marking this issue for next release.

rsjethani, 2020-12-29:
@deadprogram this is good news! 
Since both #1549 #1548 have been merged into dev, I can simply build the tinygo dev version and test my device too right?

deadprogram, 2020-12-30:
@rsjethani that is correct. Please let us know if it is working for you as expected.

rsjethani, 2020-12-31:
It seems I am having trouble installing dev version correctly. I installed tinygo and llvm as per https://tinygo.org/getting-started/linux/#source-install

```
$ tinygo version
tinygo version 0.17.0-dev linux/amd64 (using go version go1.15.6 and LLVM version 10.0.0)
```
But when I try flashing I get the following:
```
$ tinygo flash -target clue test.go 
no Go files in /home/rsj/.cache/tinygo/goroot-go1.15.6-fc1c09ad22e690a0ed027ff46974c2a43fb3839a83aaf2227dbfe90cc8c7449d-syscall/src/device/nrf
package command-line-arguments
	imports machine
	imports device/nrf
```

deadprogram, 2021-01-01:
Probably need to run `tinygo clean` to remove the old cache.

aykevl, 2021-01-01:
@rsjethani try running `make gen-device`.

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram

rsjethani, 2021-03-13:
Yes, I confirm my device works with the 0.17 release , thank you guys!