tinygo-org/tinygo #1519: panic: trying to make exported function async related to crypto/internal/randutil

sykesm, 2020-12-02:
Building code with exported functions that use the pattern found in [MaybeReadByte](https://golang.org/src/crypto/internal/randutil/randutil.go) fails with `panic: trying to make exported function async`.

While I originally ran into this while using some of the crypto packages, it's easy to observe outside of that context.

```go
package main

import (
	"crypto/rand"
	"sync"
)

var (
	closedChanOnce sync.Once
	closedChan     chan struct{}
)

//export doit
func doit() {
	closedChanOnce.Do(func() {
		closedChan = make(chan struct{})
		close(closedChan)
	})

	select {
	case <-closedChan:
		return
	case <-closedChan:
		var buf [1]byte
		rand.Reader.Read(buf[:])
	}
}

func main() {
}
```

The same error occurs if the channel is made and closed within `sync.Once#Do` without any select.
```go
package main

import (
	"sync"
)

var (
	closedChanOnce sync.Once
	closedChan     chan struct{}
)

//export doit
func doit() {
	closedChanOnce.Do(func() {
		closedChan = make(chan struct{})
		close(closedChan)
	})
}

func main() {
}
```

The same error occurs if `sync.Once` is not used and the channel is made and closed at each call:
```go
package main

import "crypto/rand"

//export doit
func doit() {
	closedChan := make(chan struct{})
	close(closedChan)

	select {
	case <-closedChan:
		return
	case <-closedChan:
		var buf [1]byte
		rand.Reader.Read(buf[:])
	}
}

func main() {
}
```

aykevl, 2020-12-07:
Yes, this is an issue at the moment and I doubt it'll get fixed. However, the error message can definitely be improved.

In short, it is not possible to export a function that might do a blocking operation, if it's done directly or indirectly through a call. However, you should be able to start a new goroutine that does this.

sykesm, 2020-12-07:
> Yes, this is an issue at the moment and I doubt it'll get fixed.

Fair enough. My ultimate goal was to be able to use the `crypto/ecdsa` package but it doesn't look like that's going to be feasible in the short term.

Regardless, thanks for feedback and all of the work that's gone into the project.

closed 2020-12-07 by sykesm