tinygo-org/tinygo #1523: STM32L432 Support

hyprnick, 2020-12-05:
Is it possible to program the STM32L432 MCU using tinygo?

aykevl, 2020-12-07:
It looks like nobody has added support for it yet, but most likely support can be added for this chip. So not right now but maybe in the future.

hyprnick, 2020-12-27:
That sounds good. Can anyone estimate the amount of work to add support?

hyprnick, 2021-01-02:
We would like to sponsor this work if anyone is available. Is there anyway to do that?

deadprogram, 2021-01-02:
@hyprnick please feel free to email or message me to see how we can help.

deadprogram, 2021-03-22:
Hello @hyprnick the STM32L432 is just about to land in the TinyGo `dev` branch:

https://github.com/tinygo-org/tinygo/pull/1710

hyprnick, 2021-03-22:
Thanks for the update @deadprogram ! That's great to hear. I will check it out.

deadprogram, 2021-05-12:
This was released with v0.18 so now closing. Thank you!

closed 2021-05-12 by deadprogram