tinygo-org/tinygo #1527: BBC micro:bit v2 (nRF52833) support

alankrantas, 2020-12-09:
I recently bought two micro:bit v2. I don't know how close is nRF52833 to nRF52840, but I can to some testing if anyone's interested to add TinyGo support for it.
https://tech.microbit.org/hardware/
https://infocenter.nordicsemi.com/pdf/nRF52833_OPS_v0.7.pdf

The Arduino Code for nF5x boards has added support (might be useful as reference):
https://github.com/sandeepmistry/arduino-nRF5

aykevl, 2020-12-12:
I don't have it myself and it's difficult to write chip support without access to the hardware. The nrf52 series is surprisingly easy in that aspect but still it'll probably be a lot of back and forth.

Can you maybe take a look at doing this? The following needs to be done:

1. Update lib/nrfx to a more recent version (v2.4.0 for example).
2. Run `make gen-device-nrf`. I discovered this results in a segfault, I haven't investigated it yet.
3. Add support for the board, with the following files:
    * `src/machine/machine_nrf52833.go`
    * `targets/nrf52833.ld`
    * `targets/nrf52833.json`
    * `src/machine/board_microbitv2.go`
    * `targets/microbitv2.json`

    You should be able to copy and modify the nrf52840 and microbit versions of those files.

If you want to do this and need any help, please let me know!

alankrantas, 2020-12-13:
I'm using my RPi 3B+ (I don't have other workable real or virtual Linux machine at home), downloaded the source and replaced ```tinygo/lib/nrfx``` to release [2.4.0](https://github.com/NordicSemiconductor/nrfx/releases/tag/v2.4.0).

This is the result of ```make gen-device-nrf```:

```
pi@raspberrypi:~/tinygo $ make gen-device-nrf
go build -o build/gen-device-svd ./tools/gen-device-svd/
./build/gen-device-svd -source=https://github.com/NordicSemiconductor/nrfx/tree/master/mdk lib/nrfx/mdk/ src/device/nrf/
lib/nrfx/mdk/nrf51.svd
lib/nrfx/mdk/nrf52.svd
lib/nrfx/mdk/nrf52805.svd
lib/nrfx/mdk/nrf52810.svd
lib/nrfx/mdk/nrf52811.svd
lib/nrfx/mdk/nrf52820.svd
lib/nrfx/mdk/nrf52833.svd
lib/nrfx/mdk/nrf52840.svd
lib/nrfx/mdk/nrf5340_application.svd
panic: runtime error: index out of range [-1]

goroutine 1 [running]:
main.readSVD(0x916150, 0x24, 0xff9723ae, 0x3b, 0x0, 0x0, 0x0)
	/home/pi/tinygo/tools/gen-device-svd/gen-device-svd.go:354 +0x27d4
main.generate(0xff9723ea, 0xd, 0xff9723f8, 0xf, 0xff9723ae, 0x3b, 0x180298, 0x8, 0x900030, 0x0)
	/home/pi/tinygo/tools/gen-device-svd/gen-device-svd.go:1063 +0x248
main.main()
	/home/pi/tinygo/tools/gen-device-svd/gen-device-svd.go:1097 +0x170
make: *** [Makefile:123: gen-device-nrf] Error 2
```

Line 354 of ```gen-device-svd.go``` is

```golang
lastReg := clusterRegisters[len(clusterRegisters)-1]
```

At some point ```clusterRegisters``` is an empty array (length 0). So I (not sure if this is ok) changed line 353 to

```golang
if dimIncrement == -1 && len(clusterRegisters) > 0 {
```

And it finished without apparent problems:

```
pi@raspberrypi:~/tinygo $ make gen-device-nrf
./build/gen-device-svd -source=https://github.com/NordicSemiconductor/nrfx/tree/master/mdk lib/nrfx/mdk/ src/device/nrf/
lib/nrfx/mdk/nrf51.svd
lib/nrfx/mdk/nrf52.svd
lib/nrfx/mdk/nrf52805.svd
lib/nrfx/mdk/nrf52810.svd
lib/nrfx/mdk/nrf52811.svd
lib/nrfx/mdk/nrf52820.svd
lib/nrfx/mdk/nrf52833.svd
lib/nrfx/mdk/nrf52840.svd
lib/nrfx/mdk/nrf5340_application.svd
lib/nrfx/mdk/nrf5340_network.svd
lib/nrfx/mdk/nrf9160.svd
GO111MODULE=off go fmt ./src/device/nrf
src/device/nrf/nrf51.go
src/device/nrf/nrf52805.go
src/device/nrf/nrf52810.go
src/device/nrf/nrf52.go
src/device/nrf/nrf52811.go
src/device/nrf/nrf52820.go
src/device/nrf/nrf52833.go
src/device/nrf/nrf52840.go
src/device/nrf/nrf5340_network.go
src/device/nrf/nrf9160.go
src/device/nrf/nrf5340_application.go
```

I'll need some time to figure out the board-related files.

alankrantas, 2020-12-15:
Question:

https://tech.microbit.org/hardware/schematic/

I assume the pins are mapped as

```
P0.00 (P0_00) = 0
...
P0.31 (P0_31) = 31
P1.00 (P1_00) = 32
...
P1.09 (P1_09) = 41 (nRF52833 has 42 pins and 82840 has 48)
```

Right?

And what should I do next to test it?

alankrantas, 2020-12-20:
Here's the support files I've tried:

[tinygo_microbitv2.zip](https://github.com/tinygo-org/tinygo/files/5720649/tinygo_microbitv2.zip)

It compiles and uploads (a ```flash.hex``` appeared in the micro:bit v2 drive and the board reboot without error) but pins are not working.

I can't read any response from ```println()``` either. Should I use ```UART0 = USB``` instead of ```UART0 = NRF_UART0```? And how do I import the USB functions? I don't understand the hierarchy and rules of machine packages and target at all.

Some other issues:

* micro:bit v2 has two I2C buses - the external one goes to the edge connector and LSM303AGR is connected to the internal one. How do we expose both to the users?
* I don't know what COMBINED_SENSOR_INT is (temperature sensor?). The capacitive touch pad on the logo is said to be on pin P1_04 according by Arduino Code for nF5x.

microbit-carlos, 2020-12-22:
> I don't know what COMBINED_SENSOR_INT is (temperature sensor?). The capacitive touch pad on the logo is said to be on pin P1_04 according by Arduino Code for nF5x.

That's a "combined sensor interrupt", an interrupt signal shared between the motion sensor and the KL27. It is not listed in this diagram, but it shows that both devices are on the same I2C bus: https://tech.microbit.org/hardware/#hardware-block-diagram.

Yes, the logo is connected to P1_04, with a 10M pull up like the edge connector pins 0, 1, and 2.

deadprogram, 2021-01-03:
I just created https://github.com/tinygo-org/tinygo/pull/1556 based on the research in this issue. It is able to run the microbit-blink and echo examples correctly.

alankrantas, 2021-01-07:
Now it's merged so I'll close this issue; thanks for everyone's help. I also tested analog input and NeoPixel driver (need to add a nRF52833 tag though) yesterday with the PR branch.

closed 2021-01-07 by alankrantas