tinygo-org/tinygo #1535: wasm memory bug/question

yuval-k, 2020-12-16:
Hello everyone! 
Than you for making tinygo wasm compatible. I've been testing it out, trying to use with envoy proxy, and I have some questions I would love your help with. I tried answering these myself by looking at the code, but I want to make sure I'm not missing anything:

I noticed that using different arguments to `--heap-size` still produces the exact same binary, with initial wasm memory set to 3 wasm-pages (i.e.: `(memory $0 3)`) I do see this argument passed in the ld flags, so I suspect that this is a bug? 

IIUC the code correctly, the initial memory is all that is available - i.e. the `memory.grow` instruction is not used to grow memory dynamically. This means that all the memory usage expected should be included in the heap-size. is that accurate?

Thank you for your reply!

asayah, 2021-01-06:
@deadprogram any update on this one?

aykevl, 2021-01-09:
Hi! Thank you for your interest in using WebAssembly. I have made a PR that implements automatically growing the heap when needed: #1567. This removes the `-heap-size` parameter entirely as it is not necessary anymore, which I believe is the proper fix (`-heap-size` was always just a workaround as the TinyGo heap previously was unable to grow in size).

djannot, 2021-01-30:
@aykevl do you know when this fix will be released ?

deadprogram, 2021-03-08:
This has been released with v0.17.0 so now closing. Thanks!

closed 2021-03-08 by deadprogram