tinygo-org/tinygo #1536: 386Linux test fails on non-x86 systems

QuLogic, 2020-12-17:
When building on [aarch64](https://koji.fedoraproject.org/koji/taskinfo?taskID=57500892), this test set fails:
```
/usr/bin/ld: i386 architecture of input file `/tmp/tinygo269308519/main.o' is incompatible with aarch64 output
/usr/bin/ld: i386 architecture of input file `/tmp/tinygo269308519/extra-0-gc_386.S.o' is incompatible with aarch64 output
/usr/bin/ld: /tmp/tinygo269308519/main.o: in function `runtime.printuint64':
/builddir/build/BUILDROOT/tinygo-0.16.0-1.fc34.aarch64/usr/lib64/tinygo/src/runtime/print.go:85: undefined reference to `__udivdi3'
collect2: error: ld returned 1 exit status
```
and many repetitions of the same.

Now, I don't have a cross-compiler or cross-glibc installed, but I also couldn't find anything in `targets/` that specified what to call (if there were anything at all; I'm not sure if linux targets use compilers or linked-to llvm). I'm not sure if I'm missing some compiler, need to patch a binary name to use the right one, or this sort of thing just isn't supported and should be skipped on other arches?

QuLogic, 2020-12-17:
Ah, I see where it's set, not in the `json` files, but in `compileopts/targets.go`. For GOARCH=386, the `CFlags` and `LDFlags` are set, but _not_ `Linker`, and it only does so on amd64. This would use regular `cc` then and not work on e.g., arm64.

In fact, is there a reason to use `cc` and cross-`gcc` instead of `clang --target=...` for linking?

aykevl, 2020-12-17:
> In fact, is there a reason to use `cc` and cross-`gcc` instead of `clang --target=...` for linking?

No, not really. That is probably a much better idea as Clang should already know where to find all these libraries.

In the long term, I would like to switch to using musl to avoid using any cross compiler. But that's unfortunately not trivial.

QuLogic, 2020-12-21:
It's not all bad, as I realized I could (and should) have been testing arm at least when it's running natively. Previously, I was just skipping it entirely as I didn't have the cross-libc.

deadprogram, 2024-12-12:
@QuLogic @aykevl is this still an issue or can it be closed?

aykevl, 2024-12-12:
It's working fine nowadays, since we don't use the system compiler/linker anymore (so no cross compilation headaches). And all tests pass on my aarch64 laptop.

closed 2024-12-12 by aykevl