tinygo-org/tinygo #1539: Reading bytes with machine.UART0.Read(buffer)

soerenbrockmann, 2020-12-22:
Hi, when sending a string like "hello" with the Arduino Serial Monitor to Arduino I expect to get the string "hello" returned to the monitor. However, with this code it prints the string in two lines like:

he
llo

```
package main

import "machine"

func main() {
	buffer := make([]byte, 100)
	for {
		n, _ := machine.UART0.Read(buffer)
		println(string(buffer))
	}
}
```

How can I fix it?

My main goal is to use an Arduino to control DC motors and Jetson Nano for AI. So the Jetson should send directions to the Arduino by UART.

deadprogram, 2020-12-27:
The code in TinyGo is returning the data read by the buffer, but faster than you expect.

You probably need to add the new data in `buffer` to another slice until you receive whatever delimiter that indicates that you have received a complete command. See the "echo" example for some code that does this sort of thing: https://github.com/tinygo-org/tinygo/blob/release/src/examples/echo/echo.go

Hope that helps!

deadprogram, 2021-09-18:
Closing due to inactivity. Please reopen if needed. Thanks!

closed 2021-09-18 by deadprogram