tinygo-org/tinygo #1541: make failed by `has no member named ‘emitZeros’`

sago35, 2020-12-22:
Make failed with the following error.
Change emitZeros(1) to EmitZeros(1) and the error goes away.

```
$ make llvm-build
(ok)

$ make
CGO_CPPFLAGS="-I/home/sago35/tinygo/tinygo/llvm-project/llvm/include -I/home/sago35/tinygo/tinygo/llvm-build/include -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/sago35/tinygo/tinygo/llvm-build/tools/clang/include -I/home/sago35/tinygo/tinygo/llvm-project/clang/include -I/home/sago35/tinygo/tinygo/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++14" CGO_LDFLAGS="/home/sago35/tinygo/tinygo/llvm-build/lib/libclang.a -L/home/sago35/tinygo/tinygo/llvm-build/lib -Wl,--start-group -lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -Wl,--end-group -lstdc++ -Wl,--start-group -llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML -Wl,--end-group -L/home/sago35/tinygo/tinygo/llvm-build/lib  -lLLVMOption -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMInterpreter -lLLVMFrontendOpenMP -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCoverage -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMXtensaDisassembler -lLLVMXtensaCodeGen -lLLVMXtensaAsmParser -lLLVMXtensaDesc -lLLVMXtensaInfo -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVCodeGen -lLLVMRISCVAsmParser -lLLVMRISCVDesc -lLLVMRISCVUtils -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMCFGuard -lLLVMGlobalISel -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMDebugInfoDWARF -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMX86Desc -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMMCDisassembler -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBinaryFormat -lLLVMX86Utils -lLLVMX86Info -lLLVMSupport -lLLVMDemangle -lrt -ldl -lpthread -lm -lstdc++ " go build -buildmode exe -o build/tinygo -tags byollvm -ldflags="-X main.gitSha1=`git rev-parse --short HEAD`" .
# github.com/tinygo-org/tinygo/builder
cc1as.cpp: In function ‘bool ExecuteAssembler(AssemblerInvocation&, clang::DiagnosticsEngine&)’:
cc1as.cpp:389:16: error: ‘class llvm::MCStreamer’ has no member named ‘emitZeros’; did you mean ‘EmitZeros’?
  389 |     Str.get()->emitZeros(1);
      |                ^~~~~~~~~
      |                EmitZeros
make: *** [Makefile:163: tinygo] エラー 2
```

ubuntu 20.04
source code: 30df912565c2ab2d69efb40a2b2859a1736a88a3

https://github.com/tinygo-org/tinygo/commit/bb27bbcb41d37d78988b571c2d0943ba808cf52d#diff-d18c1aed91d0e430e09938263e336ecd6b9ec30417530fee15ec3a13187ef7a8L386

aykevl, 2020-12-23:
It looks like you are building with LLVM 10. We recently switched to LLVM 11, so you should use that for statically linked LLVM.

deadprogram, 2020-12-27:
@sago35 you need to remove your current `llvm-project` directory, then run `make llvm-source && make llvm-build` again, now with llvm 11. I had to do the same a few days ago.

aykevl, 2020-12-27:
Also, please remove the `llvm-build` directory to be sure there are no conflicts.

sago35, 2021-01-05:
I was able to build it by executing the following
Thank you very much.

```
$ rm -rf ./llvm-build ./llvm-project
$ make llvm-build
$ make
```

closed 2021-01-05 by sago35